edgexfoundry/device-modbus-go #19: Device registration is not happening with device-modbus-go service

RoshDevi, 2019-02-14:
I have downloaded the device-modbus-go source code from edgex foundry and the compiled the code and executed command make run on the raspberry pi environment(Ubuntu aarch64 ). 

 I could see modbus micro service is running successfully.

after that i follwed the below steps( uploading device profile, addressable, device registration)
STEP 1:eg
http://localhost:48081/api/v1/deviceprofile:
[{"created":1549963354721,"modified":1549963354721,"origin":0,"description":"VI measurement","id":"56a4b46d-8ea5-460f-bb4e-c43a42267e5c","name":"test-profile","manufacturer":"XYZ","model":"IFM","labels":["modbus","industrial","voltage current"],"deviceResources":[{"description":"Breakers Current","name":"DateTime","tag":null,"properties":{"value":{"type":"Integer","readWrite":"R","minimum":"0","maximum":"65535","defaultValue":"0","size":"10","word":"2","scale":"0.001","signed":true},"units":{"type":"String","readWrite":"R","defaultValue":"milliamps"}},"attributes":{"PrimaryTable":"HOLDING_REGISTERS","StartingAddress":"5000"}}],"resources":[{"name":"DateTime","get":[{"index":"1","operation":"get","object":"DateTime","property":"value","parameter":"DateTime","resource":null,"secondary":[],"mappings":{}}],"set":null}],"commands":[{"created":1549963354716,"modified":1549963354716,"origin":0,"id":"7de226be-304b-4230-93f3-22981e779c8e","name":"DateTime","get":{"path":"/api/v1/device/{deviceId}/DateTime","responses":[{"code":"200","description":"Get the proximity sensor current reading.","expectedValues":["DateTime"]},{"code":"503","description":"service unavailable","expectedValues":[]}]},"put":{"path":null,"responses":null,"parameterNames":null}}]}]

STEP 2:
http://localhost:48081/api/v1/addressable:
[{"created":1549963524806,"modified":1549963524806,"origin":0,"id":"778db648-aa6d-4856-9e8b-d38dcb3fe1e4","name":"test-address","protocol":"HTTP","method":"GET","address":"169.254.158.144","port":502,"baseURL":"HTTP://169.254.158.144:502","url":"HTTP://169.254.158.144:502"}]

STEP 3:
device.json contains below content:
{"origin":1471806386919,"name":"test-device","description":"VI measurement","adminState":"UNLOCKED","operatingState":"ENABLED","addressable":{"name":"test-address"},"labels":["modbus","industrial","voltage current"],"location":"{lat:45.45,long:47.80}","service":{"name":"edgex-device-modbus"},"profile":{"name":"test-profile"}}

when executing below command
curl --header "Content-Type: application/json" --data @device.json -X POST http://localhost:48081/api/v1/device
iam gettin below error

**Invalid object ID: A device must be associated with a device service**

Could any one please look into this issue.

cloudxxx8, 2019-02-17:
@RoshDevi It is more like the environment issue.  Since you run the modbus service via command and other services in docker container, there might be some network or config issue you need to deal with.  Could you please help collect the following information?

1. The log of device-modbus
2. Check the name registered by device-modbus, via retrieving http://localhost:48081/api/v1/deviceservice on browser or curl

Thanks

RoshDevi, 2019-02-19:
@cloudxxx8 :  i have downloaded edgex-go sourced compiled and did make run all the microservices are running successfully, i found out the modbus-service is not available in edgex-go, i seperately download the edgex-modbus-go source compiled and did make run , modbus-service also running successfully , i have not used any docker container for building micro services.

I followed the same steps mentioned in the below link:
https://docs.edgexfoundry.org/Ch-ExamplesAddingModbusDevice.html
1)uploaded device profile
2)addressable
3)device registration but it failing with below error
**Invalid object ID: A device must be associated with a device service**

Please verify the steps which i have provided in the previous comments and help us in proceding further.

As requested log of device-modbus, i will provide you shortly

cloudxxx8, 2019-02-19:
@RoshDevi if you download the source and compile on your own, I need to know which commit exactly you are using, because the master branch is keep changing.  I have to use totally the same code to reproduce.

cloudxxx8, 2019-02-19:
Also, the following information would be helpful

* The log of device-modbus
* Check the name registered by device-modbus, via retrieving http://localhost:48081/api/v1/deviceservice on browser or curl

RoshDevi, 2019-02-21:
@cloudxxx8 I tried the docker approach of running all the micro service including the modbus-device service, all the services are up and running without any issues. But after following the below steps
1)adding device profile (success)
2)addressable (success)
3)device registration (success)

4) Iam trying to execute the below from the browser
http://localhost:48082/api/v1/device/5c6d4d749f8fc200012e90a0/command/5c6d4bf99f8fc200012e909b

but observing with below logs on the modbus-device-service
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/20 12:41:39 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/20 12:41:39 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/20 12:41:39 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/20 12:41:43 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/20 12:41:45 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/20 12:41:45 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/20 12:41:45 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/20 12:43:54 Removed device 5c6d47b89f8fc20001e01e7a
edgex-device-modbus | INFO: 2019/02/20 12:44:07 Removed device 5c6d48239f8fc20001e01e7b
edgex-device-modbus | INFO: 2019/02/20 12:46:58 Added device 5c6d4c429f8fc200012e909e
edgex-device-modbus | INFO: 2019/02/20 12:49:02 Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "": invalid syntax 
edgex-device-modbus | 
edgex-device-modbus | INFO: 2019/02/20 12:50:33 Removed device 5c6d4c429f8fc200012e909e
edgex-device-modbus | INFO: 2019/02/20 12:52:04 Added device 5c6d4d749f8fc200012e90a0
edgex-device-modbus | INFO: 2019/02/20 12:53:37 Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "": invalid syntax 
edgex-device-modbus | 
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2016/11/03 17:30:29 Check Metadata service's status ...
edgex-device-modbus | INFO: 2016/11/03 17:30:29 Check Data service's status ...
edgex-device-modbus | INFO: 2016/11/03 17:30:29 Service clients initialize successful.
edgex-device-modbus | INFO: 2016/11/03 17:30:29 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2016/11/03 17:30:30 Starting internal Scheduler
edgex-device-modbus | INFO: 2016/11/03 17:30:30 Started internal Scheduler
edgex-device-modbus | INFO: 2016/11/03 17:30:30 *Service Start() called, name=edgex-device-modbus, version=0.7.0
**edgex-device-modbus | INFO: 2016/11/03 17:34:23 Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "": invalid syntax** 

Could you please let me know what might be the problem? Thanks in advance

weichou1229, 2019-02-21:
@RoshDevi Sorry, error message not clear, device address should contain Path which described at https://docs.edgexfoundry.org/Ch-ExamplesAddingModbusDevice.html#set-up-device-service-configuration

```
     name = "Gateway address 1"
     Protocol = "TCP"
     Address = "10.211.55.6"
     Port = 502
     Path = "1"
```

RoshDevi, 2019-02-21:
@cloudxxx8  After adding the Path="1", in device address, error got vanished.
I am trying to connect modbus-server using the edgex-mobus-service
 After executing http://localhost:48082/api/v1/device/5c6d4d749f8fc200012e90a0/command/5c6d4bf99f8fc200012e909b from browser i got the below logs in modbus-device service

edgex-device-modbus | INFO: 2016/11/03 19:01:06 Modbus client create TCP connection.
edgex-device-modbus | 2016/11/03 19:01:06 http: panic serving 172.19.0.5:55398: interface conversion: interface {} is nil, not string
edgex-device-modbus | goroutine 1191 [running]:
edgex-device-modbus | net/http.(*conn).serve.func1(0x400006c960)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:1746 +0xa0
edgex-device-modbus | panic(0x31fd60, 0x40000fc8d0)
edgex-device-modbus | 	/usr/local/go/src/runtime/panic.go:513 +0x18c
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/internal/driver.createCommandInfo(0x400014cc60, 0x10, 0x400014cc70, 0x8, 0x0, 0x0, 0x400014ce50, 0x7, 0x6559f2, 0x1, ...)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/internal/driver/deviceclient.go:69 +0x358
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/internal/driver.(*Driver).handleReadCommandRequest(0x400000c580, 0x405c60, 0x40000cc000, 0x6559d1, 0x1, 0x400014d028, 0x3, 0x400014d030, 0x8, 0x400014d038, ...)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/internal/driver/driver.go:76 +0x78
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/internal/driver.(*Driver).HandleReadCommands(0x400000c580, 0x40000fa868, 0x400007e3c0, 0x1, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/internal/driver/driver.go:59 +0x3c0
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler.execReadCmd(0x40000fa800, 0x400021612c, 0x8, 0x3, 0x8, 0x1)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler/command.go:118 +0x3d0
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler.CommandHandler(0x40000fc630, 0x0, 0x0, 0x4000216100, 0x3, 0x40000fc601, 0x40000fc5d0, 0x2a85ac)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler/command.go:72 +0x264
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/controller.commandFunc(0x4055a0, 0x400033e0e0, 0x400026c800)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/controller/restfuncs.go:94 +0xa8
edgex-device-modbus | net/http.HandlerFunc.ServeHTTP(0x3aa390, 0x4055a0, 0x400033e0e0, 0x400026c800)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:1964 +0x38
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/gorilla/mux.(*Router).ServeHTTP(0x40001bf1f0, 0x4055a0, 0x400033e0e0, 0x400026c800)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/gorilla/mux/mux.go:162 +0xa4
edgex-device-modbus | net/http.serverHandler.ServeHTTP(0x400013fba0, 0x4055a0, 0x400033e0e0, 0x400026c600)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:2741 +0xb0
edgex-device-modbus | net/http.(*conn).serve(0x400006c960, 0x4059e0, 0x4000010580)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:1847 +0x4e4
edgex-device-modbus | created by net/http.(*Server).Serve
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:2851 +0x294

can you try summarizing  the above logs??

weichou1229, 2019-02-21:
@RoshDevi Could you provide your docker images version?

RoshDevi, 2019-02-21:
pi@raspberrypi:~/Desktop/Test$ sudo docker --version
Docker version 18.09.2, build 6247962
pi@raspberrypi:~/Desktop/Test$ sudo docker-compose --version
docker-compose version 1.23.2, build 1110ad0

**http://localhost:48081/api/v1/device**

[{"created":1478196864923,"modified":1478196864923,"origin":0,"description":"VI measurement","id":"581b7e809f8fc200019a9d40","name":"test-device","adminState":"UNLOCKED","operatingState":"ENABLED","addressable":{"created":1478196854258,"modified":0,"origin":0,"id":"581b7e769f8fc200019a9d3f","name":"test-address","protocol":"HTTP","method":"GET","address":"169.254.23.144","port":502,"path":"1","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://169.254.23.144:502","url":"HTTP://169.254.23.144:5021"},"lastConnected":0,"lastReported":0,"labels":[],"location":null,"service":{"created":1550658378193,"modified":1550658378193,"origin":1550658378178,"description":"","id":"5c6d2b4a9f8fc20001e71ed9","name":"edgex-device-modbus","lastConnected":0,"lastReported":0,"operatingState":"ENABLED","labels":[],"addressable":{"created":1550658378166,"modified":0,"origin":1550658378138,"id":"5c6d2b4a9f8fc20001e71ed8","name":"edgex-device-modbus","protocol":"HTTP","method":"POST","address":"edgex-device-modbus","port":49991,"path":"/api/v1/callback","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://edgex-device-modbus:49991","url":"HTTP://edgex-device-modbus:49991/api/v1/callback"},"adminState":"UNLOCKED"},"profile":{"created":1550666745419,"modified":1550666745419,"origin":0,"description":"VI measurement","id":"5c6d4bf99f8fc200012e909c","name":"test-profile","manufacturer":"Schneider Electric","model":"IFM","labels":["modbus","industrial","voltage current"],"objects":null,"deviceResources":[{"description":"Breakers Current","name":"DateTime","tag":null,"properties":{"value":{"type":"Integer","readWrite":"R","minimum":"0","maximum":"65535","defaultValue":"0","size":"10","word":"2","lsb":null,"mask":"0x00","shift":"0","scale":"0.001","offset":"0.0","base":"0","assertion":null,"signed":true,"precision":null},"units":{"type":"String","readWrite":"R","defaultValue":"milliamps"}},"attributes":{"PrimaryTable":"HOLDING_REGISTERS","StartingAddress":"5000"}}],"resources":[{"name":"DateTime","get":[{"index":"1","operation":"get","object":"DateTime","property":"value","parameter":"DateTime","resource":null,"secondary":[],"mappings":{}}],"set":null}],"commands":[{"created":1550666745417,"modified":0,"origin":0,"id":"5c6d4bf99f8fc200012e909b","name":"DateTime","get":{"path":"/api/v1/device/{deviceId}/DateTime","responses":[{"code":"200","description":"Get the proximity sensor current reading.","expectedValues":["DateTime"]},{"code":"503","description":"service unavailable","expectedValues":[]}]},"put":null}]}}]

RoshDevi, 2019-02-21:
@cloudxxx8 
sudo docker --version
Docker version 18.09.2, build 6247962
sudo docker-compose --version
docker-compose version 1.23.2, build 1110ad0

weichou1229, 2019-02-21:
@RoshDevi 
It's `sudo docker images` which can show your device-modbus image version.

RoshDevi, 2019-02-21:
@cloudxxx8 
pi@raspberrypi:~/Desktop/Test$ sudo docker images
REPOSITORY                                           TAG                 IMAGE ID            CREATED                  SIZE
edgexfoundry/docker-device-grove-c-arm64             1.0.0               55785d38d414        Less than a second ago   40.5MB
portainer/portainer                                  latest              658ac7d6c3e7        Less than a second ago   63.4MB
consul                                               1.4.0               41b2df2352f9        Less than a second ago   99.9MB
edgexfoundry/docker-support-notifications-go-arm64   0.7.1               47c398e3d25b        Less than a second ago   9.48MB
edgexfoundry/docker-support-logging-go-arm64         0.7.1               3d394bc5dd2c        Less than a second ago   9.12MB
edgexfoundry/docker-core-command-go-arm64            0.7.1               a2576dca01a3        Less than a second ago   8.25MB
edgexfoundry/docker-core-metadata-go-arm64           0.7.1               0a041685b79f        Less than a second ago   10.6MB
edgexfoundry/docker-core-data-go-arm64               0.7.1               22dcf330e911        Less than a second ago   15.6MB
edgexfoundry/docker-export-distro-go-arm64           0.7.1               8d5d01067d6f        Less than a second ago   15.7MB
edgexfoundry/docker-export-client-go-arm64           0.7.1               52f18a41340a        Less than a second ago   8.97MB
edgexfoundry/docker-core-config-seed-go-arm64        0.7.1               284424cecbd3        Less than a second ago   244MB
edgexfoundry/docker-device-modbus-go-arm64           0.7.1               3e3df9b7a276        Less than a second ago   9.33MB
edgexfoundry/docker-edgex-volume-arm64               0.8.0               e0594d99e4dc        Less than a second ago   88.6MB
edgexfoundry/docker-edgex-mongo-arm64                0.8.0               bee2e6db636f        Less than a second ago   191MB

weichou1229, 2019-02-21:
@RoshDevi 
Could you try to modify the device profile's attributes field, change uppercase to lowercase

```
"attributes":{
  "PrimaryTable":"HOLDING_REGISTERS",
  "StartingAddress":"5000"
}
```
to
```
"attributes":{
  "primaryTable":"HOLDING_REGISTERS",
  "startingAddress":"5000"
}
```

RoshDevi, 2019-02-21:
@cloudxxx8 

As per your comments i have changed attributes from upper to lower case

Modbus service logs:

edgex-device-modbus | INFO: 2016/11/03 20:54:49 Modbus client create TCP connection.
edgex-device-modbus | 2016/11/03 20:54:49 http: panic serving 172.19.0.5:38152: interface conversion: interface {} is nil, not string
edgex-device-modbus | goroutine 2588 [running]:
edgex-device-modbus | net/http.(*conn).serve.func1(0x400016a1e0)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:1746 +0xa0
edgex-device-modbus | panic(0x31fd60, 0x400005d7a0)
edgex-device-modbus | 	/usr/local/go/src/runtime/panic.go:513 +0x18c
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/internal/driver.createCommandInfo(0x400014cc60, 0x10, 0x400014cc70, 0x8, 0x0, 0x0, 0x400014ce50, 0x7, 0x6559f2, 0x1, ...)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/internal/driver/deviceclient.go:69 +0x358
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/internal/driver.(*Driver).handleReadCommandRequest(0x400000c580, 0x405c60, 0x40002bc000, 0x6559d1, 0x1, 0x400014d028, 0x3, 0x400014d030, 0x8, 0x400014d038, ...)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/internal/driver/driver.go:76 +0x78
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/internal/driver.(*Driver).HandleReadCommands(0x400000c580, 0x40002be468, 0x40001dc1e0, 0x1, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/internal/driver/driver.go:59 +0x3c0
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler.execReadCmd(0x40002be400, 0x40002662ac, 0x8, 0x3, 0x8, 0x1)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler/command.go:118 +0x3d0
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler.CommandHandler(0x4000198450, 0x0, 0x0, 0x4000266280, 0x3, 0x4000198401, 0x4000198420, 0x2a85ac)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/handler/command.go:72 +0x264
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/controller.commandFunc(0x4055a0, 0x400022c0e0, 0x40001d0600)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/edgexfoundry/device-sdk-go/internal/controller/restfuncs.go:94 +0xa8
edgex-device-modbus | net/http.HandlerFunc.ServeHTTP(0x3aa390, 0x4055a0, 0x400022c0e0, 0x40001d0600)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:1964 +0x38
edgex-device-modbus | github.com/edgexfoundry/device-modbus-go/vendor/github.com/gorilla/mux.(*Router).ServeHTTP(0x40001bf1f0, 0x4055a0, 0x400022c0e0, 0x40001d0600)
edgex-device-modbus | 	/go/src/github.com/edgexfoundry/device-modbus-go/vendor/github.com/gorilla/mux/mux.go:162 +0xa4
edgex-device-modbus | net/http.serverHandler.ServeHTTP(0x400013fba0, 0x4055a0, 0x400022c0e0, 0x40000f8600)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:2741 +0xb0
edgex-device-modbus | net/http.(*conn).serve(0x400016a1e0, 0x4059e0, 0x40000106c0)
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:1847 +0x4e4
edgex-device-modbus | created by net/http.(*Server).Serve
edgex-device-modbus | 	/usr/local/go/src/net/http/server.go:2851 +0x294

http://localhost:48081/api/v1/device

[{"created":1478206418374,"modified":1478206418374,"origin":0,"description":"VI measurement","id":"581ba3d29f8fc200019a9d44","name":"test-device","adminState":"UNLOCKED","operatingState":"ENABLED","addressable":{"created":1478206407388,"modified":0,"origin":0,"id":"581ba3c79f8fc200019a9d43","name":"test-address","protocol":"HTTP","method":"GET","address":"169.254.23.144","port":502,"path":"1","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://169.254.23.144:502","url":"HTTP://169.254.23.144:5021"},"lastConnected":0,"lastReported":0,"labels":[],"location":null,"service":{"created":1550658378193,"modified":1550658378193,"origin":1550658378178,"description":"","id":"5c6d2b4a9f8fc20001e71ed9","name":"edgex-device-modbus","lastConnected":0,"lastReported":0,"operatingState":"ENABLED","labels":[],"addressable":{"created":1550658378166,"modified":0,"origin":1550658378138,"id":"5c6d2b4a9f8fc20001e71ed8","name":"edgex-device-modbus","protocol":"HTTP","method":"POST","address":"edgex-device-modbus","port":49991,"path":"/api/v1/callback","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://edgex-device-modbus:49991","url":"HTTP://edgex-device-modbus:49991/api/v1/callback"},"adminState":"UNLOCKED"},"profile":{"created":1478206393012,"modified":1478206393012,"origin":0,"description":"VI measurement","id":"581ba3b99f8fc200019a9d42","name":"test-profile","manufacturer":"Schneider Electric","model":"IFM","labels":["modbus","industrial","voltage current"],"objects":null,"deviceResources":[{"description":"Breakers Current","name":"DateTime","tag":null,"properties":{"value":{"type":"Integer","readWrite":"R","minimum":"0","maximum":"65535","defaultValue":"0","size":"10","word":"2","lsb":null,"mask":"0x00","shift":"0","scale":"0.001","offset":"0.0","base":"0","assertion":null,"signed":true,"precision":null},"units":{"type":"String","readWrite":"R","defaultValue":"milliamps"} [... cut ...] tion":"get","object":"DateTime","property":"value","parameter":"DateTime","resource":null,"secondary":[],"mappings":{}}],"set":null}],"commands":[{"created":1478206393010,"modified":0,"origin":0,"id":"581ba3b99f8fc200019a9d41","name":"DateTime","get":{"path":"/api/v1/device/{deviceId}/DateTime","responses":[{"code":"200","description":"Get the proximity sensor current reading.","expectedValues":["DateTime"]},{"code":"503","description":"service unavailable","expectedValues":[]}]},"put":null}]}}]

weichou1229, 2019-02-21:
Do you redeploy or restart the device-modbus?

RoshDevi, 2019-02-21:
@cloudxxx8 I have not restarted the device-modbus

weichou1229, 2019-02-21:
@RoshDevi Could you restart device-modbus and then execute the command?

RoshDevi, 2019-02-21:
okay

RoshDevi, 2019-02-21:
@cloudxxx8 The modbus connection was successful with FC 3. Thanks a lot!
One clarification from our side, What is the range of device "path"? 
We are sending the query with path = 255 and it is throwing the error value out of range.

weichou1229, 2019-02-21:
What is the full error message?

RoshDevi, 2019-02-21:
The error message is : 
Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "255": value out of range

weichou1229, 2019-02-21:
Oh! It's a bug, only 0~127, should be 0~255

RoshDevi, 2019-02-21:
@cloudxxx8 , @BinduRao2018 
Queries:
Query 1:
	Our devices are the GATEWAY devices which supports fixed device path "255" , if i use device "path" as 255 then we get the below error
**Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "255": value out of range**

Query 2:
After adding deviceprofile, addressable, device registration , we are trying schedule an event so we followed below steps
1)/POST addressable
2)/POST schedule
3)/POST scheduleevent
  but we could observe in the modbus-device logs
 ==>**Schedule event callback action not implemented**

Query 3:
 edgex-ui service is not yet implemented for raspberrypi?

Query 4:
     In the modbus-device logs after 1min time out is happening , can we able to configure the timeout, if yes where we can configure the timeout?

Query 5: 
Iam planning to explore on below services in raspberry pi board. Is it possible all are supported ?
>Logging Service
>Rules Engine
>Alerts and Notification
>MQTT

RoshDevi, 2019-02-22:
@cloudxxx8 

this is my modtcp.profile.yml
name: "test-profile"
manufacturer: "xxxxxxxxxxxxxx"
model: "IFM"
description: "VI measurement"
labels: 
  - "modbus"
  - "industrial"
  - "voltage current"

deviceResources: 
      -   
        name: "DateTime"
        description: "Breakers Current"
        attributes: 
            { primaryTable: "HOLDING_REGISTERS", 
            startingAddress: "5000" }
        properties: 
            value:
                { type: "Integer", readWrite: "R", size: "10", scale: "0.001", minimum: "0", maximum: "65535", defaultValue: "0" }
            units:
                { type: "String", readWrite: "R", defaultValue: "milliamps" }
                
resources:
  -
    name: "DateTime"
    get:
      - { index: "1", operation: "get", object: "DateTime", parameter: "DateTime", property: "value" }
  
commands: 
  - 
  
    name: DateTime
    get: 
        path: "/api/v1/device/{deviceId}/DateTime"
        responses: 
          - 
            code: "200"
            description: "Get the proximity sensor current reading."
            expectedValues: ["DateTime"]
          -
            code: "503"
            description: "service unavailable"
            expectedValues: []

after executing the command below logs observed in the modbus-device service

edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2016/11/03 22:10:27 Check Metadata service's status ...
edgex-device-modbus | INFO: 2016/11/03 22:10:27 Check Data service's status ...
edgex-device-modbus | INFO: 2016/11/03 22:10:27 Service clients initialize successful.
edgex-device-modbus | INFO: 2016/11/03 22:10:28 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2016/11/03 22:10:28 Starting internal Scheduler
edgex-device-modbus | INFO: 2016/11/03 22:10:28 Started internal Scheduler
edgex-device-modbus | INFO: 2016/11/03 22:10:28 *Service Start() called, name=edgex-device-modbus, version=0.7.0
**edgex-device-modbus | ERROR: 2016/11/03 22:25:38 Schedule event callback action not implemented**
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/22 05:22:30 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/22 05:22:30 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/22 05:22:30 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/22 05:22:30 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/22 05:22:31 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/22 05:22:31 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/22 05:22:31 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/22 05:26:03 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/22 05:26:03 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 01
edgex-device-modbus | 2019/02/22 05:26:03 modbus: received 00 01 00 00 00 05 0a 03 02 01 24
edgex-device-modbus | INFO: 2019/02/22 05:26:03 Modbus client GetValue's results [1 36]
edgex-device-modbus | INFO: 2019/02/22 05:26:03 Transfer dataBytes to CommandValue(Integer, ) successful.
edgex-device-modbus | INFO: 2019/02/22 05:26:03 Read command failed. Cmd:DateTime err:return result fail, none supported value type: Integer 
edgex-device-modbus | 
edgex-device-modbus | 2019/02/22 05:27:03 modbus: closing connection due to idle timeout: 1m0.000267378s
edgex-device-modbus | INFO: 2019/02/22 05:45:03 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/22 05:45:03 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 01
edgex-device-modbus | 2019/02/22 05:45:03 modbus: received 00 01 00 00 00 05 0a 03 02 5e d0
edgex-device-modbus | INFO: 2019/02/22 05:45:03 Modbus client GetValue's results [94 208]
edgex-device-modbus | INFO: 2019/02/22 05:45:03 Transfer dataBytes to CommandValue(Integer, ) successful.
**edgex-device-modbus | INFO: 2019/02/22 05:45:03 Read command failed. Cmd:DateTime err:return result fail, none supported value type: Integer** 
edgex-device-modbus | 
edgex-device-modbus | 2019/02/22 05:46:03 modbus: closing connection due to idle timeout: 1m0.000230819s

Could you please check the logs and summarize it.

weichou1229, 2019-02-22:
In device-modbus, we support BOOL, INT16, INT32, INT64, UINT16, UINT32, UINT64, FLOAT32, FLOAT64. Please change your value type.

RoshDevi, 2019-02-22:
Read command finished, now other error , could you please review the modtcp.profile.yaml file which i shared above.

edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/22 08:22:34 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/22 08:22:34 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/22 08:22:34 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/22 08:22:35 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/22 08:22:35 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/22 08:22:35 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/22 08:22:35 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/22 08:23:00 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/22 08:23:00 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 01
edgex-device-modbus | 2019/02/22 08:23:00 modbus: received 00 01 00 00 00 05 0a 03 02 7d c3
edgex-device-modbus | INFO: 2019/02/22 08:23:00 Modbus client GetValue's results [125 195]
edgex-device-modbus | INFO: 2019/02/22 08:23:00 Transfer dataBytes to CommandValue(INT16, 32195) successful.
edgex-device-modbus | INFO: 2019/02/22 08:23:00 Read command finished. Cmd:DateTime, Origin: 1550823780182, Int16: 32195 
edgex-device-modbus | 
**edgex-device-modbus | ERROR: 2019/02/22 08:23:00 the scale 0.001 of PropertyValue cannot be parsed to int16: strconv.ParseInt: parsing "0.001": invalid syntax
edgex-device-modbus | ERROR: 2019/02/22 08:23:00 Handler - execReadCmd: CommandValue (Origin: 1550823780182, Int16: 32195) transformed failed: strconv.ParseInt: parsing "0.001": invalid syntax
edgex-device-modbus | ERROR: 2019/02/22 08:23:00 Transform failed for dev: test-device cmd: DateTime method: GET**
edgex-device-modbus | 2019/02/22 08:24:00 modbus: closing connection due to idle timeout: 1m0.000202512s

Any reference link or reference document for the editing device profile yaml file for GET, PUT, etc syntax will be appreciated.

weichou1229, 2019-02-22:
In version 0.7.1, if value type is INT and then scale should be INT too.
But this problem is fixed and merged with PR https://github.com/edgexfoundry/device-sdk-go/pull/155
You need to wait for next release.

RoshDevi, 2019-02-22:
@weichou1229 

Thanks for the valuable information, i need answer for one more query
i want to schedule an event , but i could see 
**edgex-device-modbus | ERROR: 2016/11/03 22:25:38 Schedule event callback action not implemented**

could you please comment on it.

weichou1229, 2019-02-22:
This is a design decision by the community, you need restart the device-modbus after add schedule,
otherwise you can [predefine the schedule event](https://docs.edgexfoundry.org/Ch-ExamplesAddingModbusDevice.html#set-up-device-service-configuration) and then device-modbus can create the schedule at startup.

RoshDevi, 2019-02-22:
@weichou1229 

i following the below procedure to schedule an event

STEP 1:
http://localhost:48081/api/v1/addressable
[{"created":1550658378166,"modified":0,"origin":1550658378138,"id":"5c6d2b4a9f8fc20001e71ed8","name":"edgex-device-modbus","protocol":"HTTP","method":"POST","address":"edgex-device-modbus","port":49991,"path":"/api/v1/callback","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://edgex-device-modbus:49991","url":"HTTP://edgex-device-modbus:49991/api/v1/callback"},**{"created":1478211660780,"modified":0,"origin":0,"id":"581bb84c9f8fc20001d2b3f5","name":"schedule-test","protocol":"HTTP","method":"GET","address":"edgex-device-modbus","port":49991,"path":"/api/v1/device/581bacc69f8fc200019a9d4a/DateTime","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://edgex-device-modbus:49991","url":"HTTP://edgex-device-modbus:49991/api/v1/device/581bacc69f8fc200019a9d4a/DateTime"},**{"created":1550824638800,"modified":0,"origin":0,"id":"5c6fb4be9f8fc20001c3a881","name":"test-address","protocol":"HTTP","method":"GET","address":"169.254.23.144","port":502,"path":"10","publisher":null,"user":null,"password":null,"topic":null,"baseURL":"HTTP://169.254.23.144:502","url":"HTTP://169.254.23.144:50210"}]

STEP 2:
http://localhost:48081/api/v1/schedule
[{"created":1550813396731,"modified":1550813396731,"origin":0,"id":"5c6f88d49f8fc20001c3a87a","name":"interval-test-modbus","start":null,"end":null,"frequency":"PT25S","cron":null,"runOnce":false}]

**STEP 3: I have posted the data  to below URL using curl and on console i could see the id also generated but i go check the below URL in browser  its showing Empty
http://localhost:48081/api/v1/scheduleevent
[]**

RoshDevi, 2019-02-22:
@weichou1229  event after restarting modbus-service , schedule event is not happening. I tried multiple times, could you please let me know what inputs you need for your analysis, i will provide you.

weichou1229, 2019-02-22:
I saw you just post address and schedule. Did you post scheduleEvent?

RoshDevi, 2019-02-22:
@weichou1229 
yes i have posted ScheduleEvent as well on the raspberry pi on ubutu console 

**pi@raspberrypi:~/Desktop/Test$ curl --header "Content-Type:application/json" --data @scheduleEvent.json -X POST http://localhost:48081/api/v1/scheduleevent
5c6fc6109f8fc20001c3a88b**

where scheduleEvent.json contains
{"name":"devic-event3","addressable":{"name":"schedule-test"},"schedule":"interval-test-modbus","service":"edgex-device-modbus"}

but on the browser if open
http://localhost:48081/api/v1/scheduleevent
[]

it showing empty.

weichou1229, 2019-02-22:
Could you run command `docker logs edgex-core-metadata` to watch the error logs?

cloudxxx8, 2019-02-24:
@RoshDevi did you open your browser on your Pi?  If you open the browser on your another machine, localhost might be a wrong host name

RoshDevi, 2019-02-25:
@cloudxxx8 
yes, i have opened the browser on pi iteself.

RoshDevi, 2019-02-25:
@weichou1229 
After executing docker logs edgex-core-metadata i got the below logs

pi@raspberrypi:~/Desktop/Test$ sudo docker logs edgex-core-metadata
INFO: 2016/11/03 22:06:45 Service dependencies resolved...
INFO: 2016/11/03 22:06:45 Starting edgex-core-metadata 0.7.1 
INFO: 2016/11/03 22:06:45 This is the EdgeX Core Metadata Microservice
INFO: 2016/11/03 22:06:45 Service started in: 536.868198ms
INFO: 2016/11/03 22:06:45 Listening on port: 48081
ERROR: 2016/11/03 22:22:14 Duplicate name for schedule event: Resource already exists
ERROR: 2016/11/03 22:23:56 Duplicate name for schedule event: Resource already exists
ERROR: 2016/11/03 22:24:03 Duplicate name for schedule event: Resource already exists
ERROR: 2016/11/03 22:24:35 Duplicate name for schedule event: Resource already exists
ERROR: 2016/11/03 22:24:55 Schedule event not found: mgoGetScheduleEventById Invalid Object ID 581bb34a9f8fc200019a9d52pi
ERROR: 2016/11/03 22:25:03 Schedule event not found: Item not found
INFO: 2019/02/22 05:20:00 Service dependencies resolved...
INFO: 2019/02/22 05:20:00 Starting edgex-core-metadata 0.7.1 
INFO: 2019/02/22 05:20:00 This is the EdgeX Core Metadata Microservice
INFO: 2019/02/22 05:20:00 Service started in: 753.754461ms
INFO: 2019/02/22 05:20:00 Listening on port: 48081
ERROR: 2019/02/22 05:28:08 Duplicate name for schedule event: Resource already exists
ERROR: 2019/02/22 05:28:33 Problem getting schedule: mgoGetScheduleById Invalid Object ID id
ERROR: 2019/02/22 05:29:39 Schedule not found for schedule event: Item not found
ERROR: 2019/02/22 05:30:00 Duplicate name for schedule event: Resource already exists
ERROR: 2019/02/22 05:30:28 Schedule event not found: mgoGetScheduleEventById Invalid Object ID *
ERROR: 2019/02/22 09:39:54 Schedule event not found: mgoGetScheduleEventById Invalid Object ID 5c6fc0fe9f8fc20001c3a885pi
ERROR: 2019/02/22 09:39:59 Schedule event not found: Item not found
ERROR: 2019/02/22 09:40:13 Schedule event not found: Item not found
ERROR: 2019/02/22 09:41:08 Duplicate name for schedule event: Resource already exists
ERROR: 2019/02/22 09:46:56 Schedule not found: Item not found
INFO: 2019/02/25 04:32:58 Service dependencies resolved...
INFO: 2019/02/25 04:32:58 Starting edgex-core-metadata 0.7.1 
INFO: 2019/02/25 04:32:58 This is the EdgeX Core Metadata Microservice
INFO: 2019/02/25 04:32:58 Service started in: 18.358397547s
INFO: 2019/02/25 04:32:58 Listening on port: 48081
ERROR: 2019/02/25 04:53:33 Item not found
ERROR: 2019/02/25 04:54:26 Item not found
ERROR: 2019/02/25 04:54:32 Item not found
INFO: 2019/02/25 05:45:36 Service dependencies resolved...
INFO: 2019/02/25 05:45:36 Starting edgex-core-metadata 0.7.1 
INFO: 2019/02/25 05:45:36 This is the EdgeX Core Metadata Microservice
INFO: 2019/02/25 05:45:36 Service started in: 941.730463ms
INFO: 2019/02/25 05:45:36 Listening on port: 48081
ERROR: 2019/02/25 05:56:57 Item not found

Am using mozilla firefox browser  in raspberry pi.

if i open the below URL its showing empty.

http://localhost:48081/api/v1/scheduleevent
[]

weichou1229, 2019-02-25:
@RoshDevi Here are some errors, `mgoGetScheduleEventById Invalid Object ID`, `Item not found`
Could you provide these three API request body and response?
1. schedule
2. addressable
3. schedule event

RoshDevi, 2019-02-25:
**1. schedule:**
curl --header "Content-Type:application/json" --data @schedule.json -X POST http://localhost:48081/api/v1/schedule

where schedule.json contains
**{"name":"interval-test-modbus","start":"","end":"","frequency":"PT25S"}**

Response: <Id got generated>

**2\. addressable:**
curl --header "Content-Type:application/json" --data @scheduleAddress.json -X POST http://localhost:48081/api/v1/addressable

where scheduleAddress.json contains
**{"name": "schedule-test", "protocol": "HTTP","address": "edgex-device-modbus","port":49991,"path": "/api/v1/device/5c6fb4c29f8fc20001c3a882/DateTime","method": "GET"}**

Response: <An Id got generated>

**3\. schedule event:**
curl --header "Content-Type:application/json" --data @scheduleEvent.json -X POST http://localhost:48081/api/v1/scheduleevent

where scheduleEvent.json contains
**{"name":"devic-event3","addressable":{"name":"schedule-test"},"schedule":"interval-test-modbus","service":"edgex-device-modbus"}**

Response: <An Id got generated>

For all the above there i got the response (an id got generated) when i open the browser i could see some jsons for schedule and addressable, but for schedule event its empty ( [  ] )

cloudxxx8, 2019-02-25:
@RoshDevi The log contains lots of different message from your different runs.  Could you please clean everything by the following commands, re-run and collect the log message again?  Thanks.

```
$docker-compose down
$docker volume prune
```

cloudxxx8, 2019-02-25:
please provide the log from both device-modbus and core-metadata services.

RoshDevi, 2019-02-25:
@cloudxxx8  as per you comments i executed docker commands
below are the request ones

**device-modbus service logs**
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/25 10:36:01 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/25 10:36:01 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/25 10:36:01 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/25 10:36:01 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/25 10:36:02 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 10:36:02 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 10:36:02 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/25 10:38:05 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/25 10:38:05 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 01
edgex-device-modbus | 2019/02/25 10:38:05 modbus: received 00 01 00 00 00 03 0a 83 02
edgex-device-modbus | INFO: 2019/02/25 10:38:05 Read command failed. Cmd:DateTime err:modbus: exception '2' (illegal data address), function '131' 
edgex-device-modbus | 
**edgex-device-modbus | ERROR: 2019/02/25 10:38:22 Schedule event callback action not implemented**
^CGracefully stopping... (press Ctrl+C again to force)
Stopping edgex-device-modbus ... done
pi@raspberrypi:~/Desktop/Test$ sudo docker-compose up device-modbus
edgex-files is up-to-date
edgex-core-consul is up-to-date
edgex-mongo is up-to-date
edgex-config-seed is up-to-date
edgex-support-logging is up-to-date
edgex-core-data is up-to-date
edgex-core-metadata is up-to-date
edgex-core-command is up-to-date
Starting edgex-device-modbus ... done
Attaching to edgex-device-modbus
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/25 10:39:47 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/25 10:39:47 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/25 10:39:47 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/25 10:39:47 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/25 10:39:47 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 10:39:47 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 10:39:47 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/25 10:40:12 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/25 10:40:12 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 01
edgex-device-modbus | 2019/02/25 10:40:12 modbus: received 00 01 00 00 00 03 0a 83 02
**edgex-device-modbus | INFO: 2019/02/25 10:40:12 Read command failed. Cmd:DateTime err:modbus: exception '2' (illegal data address), function '131'**

**core-metadata logs**
pi@raspberrypi:~/Desktop/Test$ sudo docker logs edgex-core-metadata
INFO: 2019/02/25 10:34:38 Service dependencies resolved...
INFO: 2019/02/25 10:34:38 Starting edgex-core-metadata 0.7.1 
INFO: 2019/02/25 10:34:38 This is the EdgeX Core Metadata Microservice
INFO: 2019/02/25 10:34:38 Service started in: 1.974526964s
INFO: 2019/02/25 10:34:38 Listening on port: 48081
pi@raspberrypi:~/Desktop/Test$

RoshDevi, 2019-02-25:
This time i was getting "Schedule event callback action not implemented" modbus logs hence i restarted the device-modbus service but still i could see no polling observed on modbus logs

RoshDevi, 2019-02-25:
**device-modbus logs**
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/25 12:00:57 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/25 12:00:57 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/25 12:00:57 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/25 12:00:57 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/25 12:00:58 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 12:00:58 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 12:00:58 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/25 12:01:51 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/25 12:01:51 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 02
edgex-device-modbus | 2019/02/25 12:01:51 modbus: received 00 01 00 00 00 07 0a 03 04 3b 2b 00 00
edgex-device-modbus | INFO: 2019/02/25 12:01:51 Modbus client GetValue's results [59 43 0 0]
edgex-device-modbus | INFO: 2019/02/25 12:01:51 Transfer dataBytes to CommandValue(FLOAT32, 0.002609253) successful.
edgex-device-modbus | INFO: 2019/02/25 12:01:51 Read command finished. Cmd:DateTime, Origin: 1551096111595, Float32: OysAAA== 
edgex-device-modbus | 
edgex-device-modbus | 2019/02/25 12:02:51 modbus: closing connection due to idle timeout: 1m0.000297463s
**edgex-device-modbus | ERROR: 2019/02/25 12:03:50 Schedule event callback action not implemented**
^CGracefully stopping... (press Ctrl+C again to force)
Stopping edgex-device-modbus ... done
pi@raspberrypi:~/Desktop/Test$ sudo docker-compose up device-modbus
edgex-files is up-to-date
edgex-core-consul is up-to-date
edgex-mongo is up-to-date
Starting edgex-config-seed ... done
edgex-support-logging is up-to-date
edgex-core-data is up-to-date
edgex-core-metadata is up-to-date
edgex-core-command is up-to-date
Starting edgex-device-modbus ... done
Attaching to edgex-device-modbus
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/25 12:05:30 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/25 12:05:30 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/25 12:05:30 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/25 12:05:30 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/25 12:05:30 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 12:05:30 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/25 12:05:30 *Service Start() called, name=edgex-device-modbus, version=0.7.0

**sudo docker logs edgex-core-metadata**
INFO: 2019/02/25 11:40:17 Service dependencies resolved...
INFO: 2019/02/25 11:40:17 Starting edgex-core-metadata 0.7.1 
INFO: 2019/02/25 11:40:17 This is the EdgeX Core Metadata Microservice
INFO: 2019/02/25 11:40:17 Service started in: 412.262976ms
INFO: 2019/02/25 11:40:17 Listening on port: 48081
pi@raspberrypi:~/Desktop/Test$ 

This time after schedule event i could observe **Schedule event callback action not implemented** in device-modbus logs, so restarted the device-modbus service and collected modbus and core-metadata logs

cloudxxx8, 2019-02-25:
@RoshDevi ok, I just found out the issue.  I believe you could see the schedule event by calling http://localhost:48081/api/v1/scheduleevent now.

The issue is caused by the addressable of schedule event.  In delhi release, we don't support the API relative path of API /api/v1/device/{device_id}/{command}, but only support /api/v1/device/name/{device_name}/{command}
{"name": "schedule-test", "protocol": "HTTP","address": "edgex-device-modbus","port":49991,"path": "/api/v1/device/5c6fb4c29f8fc20001c3a882/DateTime","method": "GET"}

Thus, please try to use /api/v1/device/name/test-device/DateTime in the addressable
Thanks

RoshDevi, 2019-02-26:
@cloudxxx8 

As per your comment i have changed the scheduleaddressable to /api/v1/device/name/test-device/DateTime, but no luck, i restart modbus service serveral times and the weird thing i am observing is always calling http://localhost:48081/api/v1/scheduleevent results
[] (empty) and on the pi console using curl command i could see while adding schedule event in response id got generated which mean schedule event is successful but on the browser its showing its empty.

RoshDevi, 2019-02-26:
@weichou1229
@cloudxxx8
@BinduRao2018

**device modbus logs:**
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/26 06:11:01 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/26 06:11:01 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/26 06:11:01 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/26 06:11:02 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/26 06:11:02 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/26 06:11:02 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/26 06:11:02 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/26 06:14:34 Removed device 5c73cc399f8fc20001f72357
edgex-device-modbus | INFO: 2019/02/26 06:16:50 Added device 5c74d9d29f8fc2000106ec8c
edgex-device-modbus | ERROR: 2019/02/26 06:17:14 Schedule event callback action not implemented
^CGracefully stopping... (press Ctrl+C again to force)
Stopping edgex-device-modbus ... done
pi@raspberrypi:~/Desktop/Test$ sudo docker-compose up device-modbus
edgex-files is up-to-date
edgex-core-consul is up-to-date
edgex-mongo is up-to-date
Starting edgex-config-seed ... done
edgex-support-logging is up-to-date
edgex-core-data is up-to-date
edgex-core-metadata is up-to-date
edgex-core-command is up-to-date
Starting edgex-device-modbus ... done
Attaching to edgex-device-modbus
edgex-device-modbus | Init: useRegistry: true profile: docker confDir: /res
edgex-device-modbus | Check registry is up... http://edgex-core-consul:8500
edgex-device-modbus | Register the Service ...
edgex-device-modbus | Register the Health Check ...
edgex-device-modbus | Register in registry...
edgex-device-modbus | Calling service.Start.
edgex-device-modbus | EnableRemote is false, using local log file
edgex-device-modbus | INFO: 2019/02/26 06:18:31 Check Metadata service's status ...
edgex-device-modbus | INFO: 2019/02/26 06:18:31 Check Data service's status ...
edgex-device-modbus | INFO: 2019/02/26 06:18:31 Service clients initialize successful.
edgex-device-modbus | INFO: 2019/02/26 06:18:32 Device Service edgex-device-modbus exists
edgex-device-modbus | INFO: 2019/02/26 06:18:32 Starting internal Scheduler
edgex-device-modbus | INFO: 2019/02/26 06:18:32 Started internal Scheduler
edgex-device-modbus | INFO: 2019/02/26 06:18:32 *Service Start() called, name=edgex-device-modbus, version=0.7.0
edgex-device-modbus | INFO: 2019/02/26 06:19:27 Modbus client create TCP connection.
edgex-device-modbus | 2019/02/26 06:19:27 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 02
edgex-device-modbus | 2019/02/26 06:19:27 modbus: received 00 01 00 00 00 07 0a 03 04 0c 1e 00 00
edgex-device-modbus | INFO: 2019/02/26 06:19:27 Modbus client GetValue's results [12 30 0 0]
edgex-device-modbus | INFO: 2019/02/26 06:19:27 Transfer dataBytes to CommandValue(FLOAT32, 1.2171877e-31) successful.
edgex-device-modbus | INFO: 2019/02/26 06:19:27 Read command finished. Cmd:DateTime, Origin: 1551161967085, Float32: DB4AAA== 
edgex-device-modbus | 
edgex-device-modbus | 2019/02/26 06:20:27 modbus: closing connection due to idle timeout: 1m0.000213822s

**sudo docker logs edgex-core-metadata**
INFO: 2019/02/26 06:10:13 Service dependencies resolved...
INFO: 2019/02/26 06:10:16 Starting edgex-core-metadata 0.7.1 
INFO: 2019/02/26 06:10:16 This is the EdgeX Core Metadata Microservice
INFO: 2019/02/26 06:10:16 Service started in: 4.65142883s
INFO: 2019/02/26 06:10:16 Listening on port: 48081
ERROR: 2019/02/26 06:14:03 Data integrity issue: attempt to delete addressable still in use by a device or device service
ERROR: 2019/02/26 06:16:30 Schedule event not found: Item not found

**http://localhost:48081/api/v1/deviceprofile:**

[{"created":1551092776104,"modified":1551092776104,"origin":0,"description":"VI measurement","id":"5c73cc289f8fc20001f72355","name":"test-profile","manufacturer":"Schneider Electric","model":"IFM","labels":["modbus","industrial","voltage current"],"objects":null,"deviceResources":[{"description":"Breakers Current","name":"DateTime","tag":null,"properties":{"value":{"type":"FLOAT32","readWrite":"R","minimum":"0","maximum":"65535","defaultValue":"0","size":"10","word":"2","lsb":null,"mask":"0x00","shift":"0","scale":"0.001","offset":"0.0","base":"0","assertion":null,"signed":true,"precision":null},"units":{"type":"String","readWrite":"R","defaultValue":"milliamps"}},"attributes":{"primaryTable":"HOLDING_REGISTERS","startingAddress":"5000"}}],"resources":[{"name":"DateTime","get":[{"index":"1","operation":"get","object":"DateTime","property":"value","parameter":"DateTime","resource":null,"secondary":[],"mappings":{}}],"set":null}],"commands":[{"created":1551092776101,"modified":0,"origin":0,"id":"5c73cc289f8fc20001f72354","name":"DateTime","get":{"path":"/api/v1/device/{deviceId}/DateTime","responses":[{"code":"200","description":"Get the proximity sensor current reading.","expectedValues":["DateTime"]},{"code":"503","description":"service unavailable","expectedValues":[]}]},"put":null}]}]

[... cut ...] }}]

**http://localhost:48081/api/v1/schedule:**
[{"created":1551161826343,"modified":1551161826343,"origin":0,"id":"5c74d9e29f8fc2000106ec8e","name":"interval-test-modbus","start":null,"end":null,"frequency":"PT25S","cron":null,"runOnce":false}]

**http://localhost:48081/api/v1/scheduleevent**
[]

Am properly following all the steps for scheduling mechanism ,but no proper result could any one of you check all the logs and request API's and let me know what will be the issue.

Thanks in advance.

cloudxxx8, 2019-02-26:
@RoshDevi it looks like your command works.  Did you see the reading keep generating?
http://localhost:48080/api/v1/reading

> edgex-device-modbus | INFO: 2019/02/26 06:19:27 Modbus client create TCP connection.
> edgex-device-modbus | 2019/02/26 06:19:27 modbus: sending 00 01 00 00 00 06 0a 03 13 87 00 02
> edgex-device-modbus [...]

RoshDevi, 2019-02-26:
@cloudxxx8 

http://localhost:48080/api/v1/reading

[{"id":"5c73cd4b0e36080001e7c660","pushed":0,"created":1551093067412,"origin":1551093067405,"modified":0,"device":"test-device","name":"DateTime","value":"SmuFHw=="},{"id":"5c73d5aa0e3608000199e5d0","pushed":0,"created":1551095210415,"origin":1551095210387,"modified":0,"device":"test-device","name":"DateTime","value":"FGJN1A=="},{"id":"5c73d64b0e3608000199e5d2","pushed":0,"created":1551095371410,"origin":1551095371390,"modified":0,"device":"test-device","name":"DateTime","value":"AAS8aw=="},{"id":"5c73d8290e3608000199e5d4","pushed":0,"created":1551095849774,"origin":1551095849760,"modified":0,"device":"test-device","name":"DateTime","value":"ABP3zw=="},{"id":"5c73d8370e3608000199e5d6","pushed":0,"created":1551095863153,"origin":1551095863149,"modified":0,"device":"test-device","name":"DateTime","value":"QkKPXQ=="},{"id":"5c73d92f0e3608000199e5d8","pushed":0,"created":1551096111603,"origin":1551096111595,"modified":0,"device":"test-device","name":"DateTime","value":"Ni8aoA=="},{"id":"5c74d2e00e360800014ccef1","pushed":0,"created":1551160032005,"origin":1551160030566,"modified":0,"device":"test-device","name":"DateTime","value":"EECDEw=="},{"id":"5c74d3550e360800014ccef3","pushed":0,"created":1551160149854,"origin":1551160149839,"modified":0,"device":"test-device","name":"DateTime","value":"cYGJOA=="},{"id":"5c74d6b60e360800014ccef5","pushed":0,"created":1551161014115,"origin":1551161014108,"modified":0,"device":"test-device","name":"DateTime","value":"f/4AAA=="},{"id":"5c74da6f0e36080001dc6376","pushed":0,"created":1551161967100,"origin":1551161967085,"modified":0,"device":"test-device","name":"DateTime","value":"ByHKwQ=="}]

RoshDevi, 2019-02-26:
every time i execute the command
http://localhost:48082/api/v1/device/5c74d9d29f8fc2000106ec8c/command/5c73cc289f8fc20001f72354

its getting read by http://localhost:48080/api/v1/reading

but scheduling is not happening right? what ever its observed is manual thing not the scheduling mechanism

cloudxxx8, 2019-02-26:
right, it doesn't happen, but other functions work correctly.
One issue we found is that the version of core-metadata and device-modbus is not matched.
Could you please modify your docker-compose file to make all the images the same, no matter 0.7.0 or 0.7.1?  The schedule event id format is changed between these two version.
Thanks

RoshDevi, 2019-02-26:
pi@raspberrypi:~/Desktop/Test$ sudo docker images
REPOSITORY TAG IMAGE ID CREATED SIZE
edgexfoundry/docker-device-grove-c-arm64 1.0.0 55785d38d414 Less than a second ago 40.5MB
portainer/portainer latest 658ac7d6c3e7 Less than a second ago 63.4MB
consul 1.4.0 41b2df2352f9 Less than a second ago 99.9MB
edgexfoundry/docker-support-notifications-go-arm64 0.7.1 47c398e3d25b Less than a second ago 9.48MB
edgexfoundry/docker-support-logging-go-arm64 0.7.1 3d394bc5dd2c Less than a second ago 9.12MB
edgexfoundry/docker-core-command-go-arm64 0.7.1 a2576dca01a3 Less than a second ago 8.25MB
edgexfoundry/docker-core-metadata-go-arm64 0.7.1 0a041685b79f Less than a second ago 10.6MB
edgexfoundry/docker-core-data-go-arm64 0.7.1 22dcf330e911 Less than a second ago 15.6MB
edgexfoundry/docker-export-distro-go-arm64 0.7.1 8d5d01067d6f Less than a second ago 15.7MB
edgexfoundry/docker-export-client-go-arm64 0.7.1 52f18a41340a Less than a second ago 8.97MB
edgexfoundry/docker-core-config-seed-go-arm64 0.7.1 284424cecbd3 Less than a second ago 244MB
edgexfoundry/docker-device-modbus-go-arm64 0.7.1 3e3df9b7a276 Less than a second ago 9.33MB
edgexfoundry/docker-edgex-volume-arm64 0.8.0 e0594d99e4dc Less than a second ago 88.6MB
edgexfoundry/docker-edgex-mongo-arm64 0.8.0 bee2e6db636f Less than a second ago 191MB

from the above we could see already all are 0.7.1 version only (meta and device-modbus is of same version) except, volume and mongo which is 0.8.0, but when i tried to change the version of volume and mongo to 0.7.1 its resulted in 
for volume / mongo manifest for edgexfoundry/docker-edgex-volume/mongo-arm64:0/0.7.1 not found

cloudxxx8, 2019-02-26:
sorry, that's our mistake.  the version number on device-modbus log is wrong.

RoshDevi, 2019-02-26:
@weichou1229
@cloudxxx8
@BinduRao2018

can i assume scheduling will not working in the current release??  If its getting delayed then i want to explore more on below 

Objective:
My aim to fetch the data from modbus server and push it in to mongo db and based on some kind of analytics/computing i will push the data to cloud.

In order to achieve above what are all edgex services i need to explore
could any one share me link/documents/ any help regarding the above aim will be highly appreciated.

weichou1229, 2019-02-27:
I don't have arm64 for testing, but scheduling is work on my x86_64 machine and community is work on next release now. For more tutorial guide, You can refer [doc](https://docs.edgexfoundry.org/).

RoshDevi, 2019-02-28:
@weichou1229 
@cloudxxx8 
@BinduRao2018 
Could you direct me to people who are working on the arm64. I will try to coordinate and continue my conversation with them.

RoshDevi, 2019-02-28:
@weichou1229
@cloudxxx8
@BinduRao2018

Hi All,

i reflashed entire image , installed docker , executed docker up command, now this time sheduling mechanism is working as expected without any issues in raspberry pi.

Thanks you all for your continuous support.

RoshDevi, 2019-02-28:
@weichou1229
@cloudxxx8
@BinduRao2018
Now Scheduling mechanism is working, now i want to store the polled data into mongo db. How can we do that?

cloudxxx8, 2019-03-03:
@RoshDevi I am glad you make it work.  Please close this issue because we have confirm this is not the issue of device-modbus.  Also, I not sure what your question is.  The polled data should be stored into MongoDB by default.
Instead, you could raise your question on slack channels, and I believe you could get more support their.
https://edgexfoundry.slack.com

closed 2019-03-04 by RoshDevi