edgexfoundry/device-modbus-go #23: Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "255": value out of range

RoshDevi, 2019-03-01:
Our devices are the GATEWAY devices which supports fixed device path "255" , if i use device "path" as 255 in addressable then we get the below error

**Read command createConnectionInfo failed. err:strconv.ParseInt: parsing "255": value out of range**

after discussion with edgex team member i came to know that its bug in implementation of edgex
 only 0-127 supported currently , it should be 0-255.

Raising this issue  as reference to close it in the upcoming edgex releases

cloudxxx8, 2019-03-03:
@RoshDevi Thanks for reporting this issue.  It is really a bug.  We should support larger integer for the unit id of Modbus device.

@weichou1229 please replace this line with ParseUint function and add unit test for path is 0 and 255.
https://github.com/edgexfoundry/device-modbus-go/blob/master/internal/driver/deviceclient.go#L53

weichou1229, 2019-03-04:
PR is ready for review.

closed 2019-03-04 by SteveOss

RoshDevi, 2019-03-05:
@weichou1229 

in docker-compose.yaml file currently am using below version of device-modbus
**image: edgexfoundry/docker-device-modbus-go-arm64:0.7.1**

but if the device path issue is fixed then from my side which image i need to use?

cloudxxx8, 2019-03-05:
@RoshDevi please try
nexus3.edgexfoundry.org:10004/docker-device-modbus-go-arm64:1.0.0

RoshDevi, 2019-03-06:
@cloudxxx8 
i placed nexus3.edgexfoundry.org:10004/docker-device-modbus-go-arm64:1.0.0 in docker-compose.yaml file but when i docker-compose pull i get the below error

**ERROR: for device-modbus Get https://nexus3.edgefoundry.org:10004/v2/:dial tcp:lookup nexus3.edgefoundry.org on 192.168.43.1:53 no such host**

RoshDevi, 2019-03-06:
@cloudxxx8 
@weichou1229 

below 
# Copyright 2017 Dell Inc. All rights reserved.
name: "test-profile"
manufacturer: "Schneider Electric"
model: "IFM"
description: "VI measurement"
labels: 
  - "modbus"
  - "industrial"
  - "voltage current"

deviceResources: 
      -   
        name: "DateTime"
        description: "Breakers Current"
        attributes: 
            { primaryTable: "HOLDING_REGISTERS", 
            startingAddress: "5000" }
        properties: 
            value:
                { type: "INT32", readWrite: "R", size: "10", scale: "1", minimum: "0", maximum: "65535", defaultValue: "0" }
            units:
                { type: "String", readWrite: "R", defaultValue: "milliamps" }
                
resources:
  -
    name: "DateTime"
    get:
      - { index: "1", operation: "get", object: "DateTime", parameter: "DateTime", property: "value" }
  
commands: 
  - 
  
    name: DateTime
    get: 
        path: "/api/v1/device/{deviceId}/DateTime"
        responses: 
          - 
            code: "200"
            description: "Get the proximity sensor current reading."
            expectedValues: ["DateTime"]
          -
            code: "503"
            description: "service unavailable"
            expectedValues: []

**i want to place PUT functionality as well in the device profile, could you please provide me the PUT syntax API**

weichou1229, 2019-03-06:
@RoshDevi please try `docker pull  nexus3.edgexfoundry.org:10004/docker-device-modbus-go-arm64:1.0.0 `
If command work, it possible that you set the wrong path in docker compose file.

Put command can follow the [doc](https://docs.edgexfoundry.org/Ch-ExamplesAddingModbusDevice.html) and [code](https://github.com/edgexfoundry/device-modbus-go/blob/master/cmd/res/example/modbus.test.device.profile.yml#L172)

RoshDevi, 2019-03-06:
nexus3.edgexfoundry.org:10004/docker-device-modbus-go-arm64:1.0.0

i tried using the above image but device-modbus is getting exited. please look into it.

weichou1229, 2019-03-06:
Could you please show the error logs?

PR #24 (2019-03-04): Replace ParseInt with ParseUint to fix unitID out of rage
Fix #23