edgexfoundry/device-modbus-go #27: Schedule Event is working only for the first time of the fresh boot of raspberry pi, if we re reschedule the event then its not working, I feel its an issue. Could you please confirm any one confirm

RoshDevi, 2019-03-05:
I feel its an issue. Could you please confirm any one confirm

cloudxxx8, 2019-03-05:
@RoshDevi yes, it is an issue, but not a Modbus issue.
The problem is that the ScheduleEvent API doesn't work.  Could you please move this issue to edgex-go repository or slack?  Thanks.
Also, please mention which version you are using.

RoshDevi, 2019-03-05:
Sure i will issue in edgex-go

@cloudxxx8 

am editing the device profile with PUT functionality
eg:

put:
      path: "/api/v1/device/{deviceId}/DateTime"
      parameterNames: [ "Min_Int64","Max_Int64"]

while adding the profile using curl command its stating
yaml:line (where the parameterNames object is present) :found character that cannot start any token 

is anything wrong in the syntax?

cloudxxx8, 2019-03-05:
I am not sure which API you are using, but profile update doesn't support YAML.  We can only create Device Profile in YAML, not update.

RoshDevi, 2019-03-06:
@cloudxxx8 
@weichou1229 

**below is the device profile am using currently to get the data from the modbus server to edgexdevice-modbus client now i want send data from client to server what changes i have to make in the device profile?**

****************************************************************
# Copyright 2017 Dell Inc. All rights reserved.
name: "test-profile"
manufacturer: "Schneider Electric"
model: "IFM"
description: "VI measurement"
labels: 
  - "modbus"
  - "industrial"
  - "voltage current"

deviceResources: 
      -   
        name: "DateTime"
        description: "Breakers Current"
        attributes: 
            { primaryTable: "HOLDING_REGISTERS", 
            startingAddress: "5000" }
        properties: 
            value:
                { type: "INT32", readWrite: "R", size: "10", scale: "1", minimum: "0", maximum: "65535", defaultValue: "0" }
            units:
                { type: "String", readWrite: "R", defaultValue: "milliamps" }
                
resources:
  -
    name: "DateTime"
    get:
      - { index: "1", operation: "get", object: "DateTime", parameter: "DateTime", property: "value" }
  
commands: 
  - 
  
    name: DateTime
    get: 
        path: "/api/v1/device/{deviceId}/DateTime"
        responses: 
          - 
            code: "200"
            description: "Get the proximity sensor current reading."
            expectedValues: ["DateTime"]
          -
            code: "503"
            description: "service unavailable"
            expectedValues: []

****************************************************************

cloudxxx8, 2019-03-08:
@RoshDevi sorry, I don't really understand your question, and it is not the topic of this.  if you think it is a bug, please open another issue and describe your reproduce steps, expected result, and actual result.  For the schedule event issue, please close this one and open a new one on edgex-go with the image name you are using.

closed 2019-03-08 by RoshDevi

cloudxxx8, 2019-03-08:
@RoshDevi thanks, please tag me when you create the new issue, and I will help coordinate it