edgexfoundry/device-modbus-go #61: startingAddress shouldn't be decremented

drensber, 2019-06-19:
The device-modbus-go implementation currently decrements the value of 'startingAddress' that is read from the device profile by 1. 

This is confusing for a number of reasons:
  - It is different from the behavior of the original Java device-modbus implementation.
  - It means that the numbering for resources of most devices will start at "2", which seems kind of odd.
  - It makes the values declared in the device profile incongruent with the values used in the serial protocol. 

I realize that changing this would force people who are already using device-modus-go to change their device profiles, but right now, I suspect that's a fairly small number of people, and I think it's better to get these things right before usage is so widespread that it's impossible to change.

weichou1229, 2019-06-20:
In modbus protocol, 1~9999 used to set up the register number and modbus driver will minus 1 as actual data address. That's why we decrease the number.

Is it okay with you?

http://www.simplymodbus.ca/FAQ.htm#Stored

Coil/Register Numbers | Data Addresses | Type | Table Name
-- | -- | -- | --
1-9999 | 0000 to 270E | Read-Write | Discrete Output Coils
10001-19999 | 0000 to 270E | Read-Only | Discrete Input Contacts
30001-39999 | 0000 to 270E | Read-Only | Analog Input Registers
40001-49999 | 0000 to 270E | Read-Write | Analog Output Holding Registers

cloudxxx8, 2019-06-20:
we decided to use 1-based address instead of 0-based address for user friendly.
This change will break any existing profile and document.
There are some device user manuals use 1-based address and some use 0-based address.

drensber, 2019-06-20:
Hi Cloud,

As I understand it, the term "number" is used to refer to the 1-based
entity (which also includes type information) and "address" is used to
refer to the 0-based entity. The link that Bruce sent confirms that, as
does this:
https://en.wikipedia.org/wiki/Modbus#Coil,_discrete_input,_input_register,_holding_register_numbers_and_addresses

I think the fact that we're calling using the word "address" to refer to a
"number" (but without the type information) would confuse most people who
are familiar with Modbus but new to EdgeX. I realize that this would break
compatibility with device profiles that people used in Delhi, but it seems
like this would be the right time to correct something like this, since
people are already expecting Edinburgh to break compatibility with their
existing setup.

--Dave

cloudxxx8, 2019-06-21:
@drensber It's still too late to make this change into Edinburgh release, because the code freeze date was May 28th, and the release date is June 20th.
Also, most user manual we saw is represented in 1-based address.  Users could make their own profile according to the device user manual directly.

By the way, this is a reference implementation.  Users could always make their own Modbus Device Service to fit their needs.

wlgardner, 2020-08-15:
You could always add a bool value that represents “zero based addressing”. I’ve seen this in many modbus implementations. If true then -1 else do nothing.

iain-anderson, 2020-08-19:
Adding a config for zero-based / one-based is ok but may lead to problems when using a range of profiles with different assumptions?

wlgardner, 2020-08-29:
Hi,
I made a fork and implemented some changes, that I think may handle this. If anyone would like to have a look and give an opinion, I would appreciate it. I am not a professional programmer, and I'm still wrapping my head around it. Changes were made to driver.go, deviceClient.go, deciveClient_test.go and config.go

Best Regards

cloudxxx8, 2020-08-30:
you might need to provide the URL of your commit, and people can review your change

wlgardner, 2020-08-30:
Sorry about that. https://github.com/wlgardner/device-modbus-go

weichou1229, 2020-08-31:
Hi @wlgardner, 
I saw there are two commits in the https://github.com/wlgardner/device-modbus-go but you need to open a new branch and create a PR on https://github.com/edgexfoundry/device-modbus-go.

You can follow below link for more detail contribute guideline.
https://wiki.edgexfoundry.org/display/FA/Committing+Code+Guidelines

wlgardner, 2020-09-04:
Hi @weichou1229, I've opened a new branch, etc..created a pull request. This is the first time contributing to a project, so I am not very good with git. I'm not sure what the issues are, so if you can have a look and advise when you are free i'd appreciate it. Best regards.

weichou1229, 2020-09-05:
No problem, I will do that.

iain-anderson, 2020-10-28:
Codefreeze for Hanoi is here; bumping to Ireland

iain-anderson, 2021-01-13:
As per WG discussions 4th & 11th Jan this should be implemented as per the OP, ie startingAddress should not be decremented. This needs to be clearly documented and any example profiles updated.

cloudxxx8, 2021-04-25:
@weichou1229 we are going to support uint in the attributes of profile. you can check if the startingAddress is string, means it is v1 profile.  In v2, we would use uint for startingAddress.
v1 keeps the same behavior (decremented), and v2 removes the decremented logic.

iain-anderson, 2021-04-26:
Instead, reject string-typed attribute so that existing profiles must be updated

commit 1219b9d1 (2021-05-11): feat: Use zero-based startingAddress

- Use zero-based startingAddress
- Use go v1.16
- Parse deviceResource from interface{}
- Upgrade go-mod-core-contracts lib to v2.0.0-dev.83
- Upgrade device-sdk-go lib to v2.0.0-dev.51

Close #61
closed 2021-05-11 by cloudxxx8