edgexfoundry/device-modbus-go #72: Modbus example is not working

MandipJ, 2019-08-14:
Hi,

I have been trying to bring up the modbus edgex device service by following the example below:
https://docs.edgexfoundry.org/Ch-ExamplesAddingModbusDevice.html\

I have set up ModebusPal, made the required changes in the configuration.toml file.

When trying to run the command **http://localhost:48082/api/v1/device/98de8650-85a4-4591-9f29-  c4058bca6afb/command/ac024bff-1a0a-44c9-8ec0-b03a1c006e52** ,it shows the following error:

Handler - execReadCmd: error for Device: Power meter 1 cmd: Configuration, dial tcp 0.0.0.0:1502: connect: connection refused /api/v1/device/98de8650-85a4-4591-9f29-c4058bca6afb/Configuration

Attached the docker-compose.yml, configuration.toml and device profile being used.
[files.zip](https://github.com/edgexfoundry/device-modbus-go/files/3501037/files.zip)

weichou1229, 2019-08-14:
0.0.0.0 is invalid, you should give available IP like 192.168.0.123.

MandipJ, 2019-08-19:
@weichou1229  Thanks for the help! Device service is able to connect to the Modbuspal.

But there is one more error I am getting while writing the values to the Register.
**Writing:**
[
	{"DemandWindowSize": "150"},
	{"LineFrequency": "60"}
]

**Error:**
Handler - execWriteCmd: Put parameters parsing failed: [
	{"DemandWindowSize": "150"},
	{"LineFrequency": "60"}
] /api/v1/device/291e4040-4ac0-4f6f-aaeb-154be1c5a104/Configuration

weichou1229, 2019-08-19:
In edinburgh release, the request body should be:
```
{
  "DemandWindowSize":"150",
  "LineFrequency":"60"
}
```

MandipJ, 2019-08-19:
Thanks! Its working.

closed 2019-08-19 by MandipJ