edgexfoundry/device-modbus-go #75: modbus run mian.go

yannuo0802, 2019-08-21:
# github.com/edgexfoundry/device-modbus-go/internal/driver
internal/driver/driver.go:233:2: cannot use driver (type *Driver) as type "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver in return argument:
*Driver does not implement "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver (missing AddDevice method)

https://github.com/edgexfoundry/device-modbus-go/blob/739f7eff6e149c6b3d458393f9a9feebe37a89e8/internal/driver/driver.go#L229

weichou1229, 2019-08-21:
@yannuo0802 Is it edinburgh release? Could you describe how to reproduce this issue?

yannuo0802, 2019-08-21:
> @yannuo0802 Is it edinburgh release? Could you describe how to reproduce this issue?

@weichou1229 yes,it is edinburgh release
Run the modbus-go project in the development environment and run the mian.go file. Tip: /edgexfoundry/device-modbus-go/interna/driver/driver.go:233:2: cannot use driver (type *Driver) as type "github .com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver in return argument:
*Driver does not implement "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver (missing AddDevice method)

weichou1229, 2019-08-21:
Do you checkout the edinburgh branch. I got this issue at master branch.

yannuo0802, 2019-08-21:
> Do you checkout the edinburgh branch? I got this issue at master branch.

@weichou1229 
I have tried it, I have encountered the same problem, now I have a problem with configuring the modbus device, so I want to check it in the development environment debug, but the modbus-go local development environment starts to report an error.
[image]

weichou1229, 2019-08-21:
Sorry, I retry to reproduce your error, and the error only occurs when I run `go get github.com/edgexfoundry/device-sdk-go@master` to get the newest SDK.
```
$ git clone https://github.com/edgexfoundry/device-modbus-go.git
Cloning into 'device-modbus-go'...
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 326 (delta 0), reused 5 (delta 0), pack-reused 319
Receiving objects: 100% (326/326), 99.79 KiB | 164.00 KiB/s, done.
Resolving deltas: 100% (146/146), done.
$ cd device-modbus-go/
$ go test ./...
?   	github.com/edgexfoundry/device-modbus-go	[no test files]
?   	github.com/edgexfoundry/device-modbus-go/cmd	[no test files]
ok  	github.com/edgexfoundry/device-modbus-go/internal/driver	0.019s
?   	github.com/edgexfoundry/device-modbus-go/mock	[no test files]

$ git checkout edinburgh
Branch 'edinburgh' set up to track remote branch 'edinburgh' from 'origin'.
Switched to a new branch 'edinburgh'
$ go test ./...
?   	github.com/edgexfoundry/device-modbus-go	[no test files]
?   	github.com/edgexfoundry/device-modbus-go/cmd	[no test files]
ok  	github.com/edgexfoundry/device-modbus-go/internal/driver	0.019s
?   	github.com/edgexfoundry/device-modbus-go/mock	[no test files]

$ go get github.com/edgexfoundry/device-sdk-go@master
go: finding github.com/edgexfoundry/device-sdk-go master
$ go test ./...
?   	github.com/edgexfoundry/device-modbus-go	[no test files]
# github.com/edgexfoundry/device-modbus-go/internal/driver
internal/driver/driver.go:233:2: cannot use driver (type *Driver) as type "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver in return argument:
	*Driver does not implement "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver (missing AddDevice method)
# github.com/edgexfoundry/device-modbus-go/internal/driver [github.com/edgexfoundry/device-modbus-go/internal/driver.test]
internal/driver/driver.go:233:2: cannot use driver (type *Driver) as type "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver in return argument:
	*Driver does not implement "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver (missing AddDevice method)
FAIL	github.com/edgexfoundry/device-modbus-go/internal/driver [build failed]
```

And the exception '2' (illegal data address) you got means the address is unavailable on the device. You can find the description on the [doc](http://www.simplymodbus.ca/exceptions.htm)

yannuo0802, 2019-08-21:
> github.com/edgexfoundry/device-sdk-go

@weichou1229 
Command mode, I can also use this, but I am using the idea debug mode runtime error.
##
nternal/driver/driver.go:233:2: cannot use driver (type *Driver) as type "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver in return argument:
	*Driver does not implement "github.com/edgexfoundry/device-sdk-go/pkg/models".ProtocolDriver (missing AddDevice method)
##

weichou1229, 2019-08-21:
My IDE is GoLand and it works fine, please check the SDK version is v1.0.0 in your go.mod
`github.com/edgexfoundry/device-sdk-go v1.0.0`

closed 2019-08-22 by yannuo0802