edgexfoundry/device-modbus-go #77: Device Profile Ineffective

zhanglu0704, 2019-08-21:
I found the following problems when using docker image : 
    edgexfoundry/docker-device-modbus-go-arm 64:1.0.0 .
[image]
    I configured porfile like：
```
        name: "Temperature"
        description: " 0.1°C"
        attributes:
            { primaryTable: "INPUT_REGISTERS", startingAddress: "0/1/30000/30001/" }
        properties:
            value:
                { type: "FLOAT32", readWrite: "R", scale: 0.1, floatEncoding: "Base64"}
            units:
                { type: "String", readWrite: "R", defaultValue: "°C"}
```
        The message I need to send is：00  01  00  00  00  06  00  04  00  00  00  02
        In fact I tried:0，1， 30000 ， 30001，because I don't know relative address or 
    absolute address witch one I shoud use，and I find : 
    "startingAddress = startingAddress - 1"  in deviceclient.go.
        Nevertheless, there is no change in the content sent. Still sending：
    00 01 00 00 00 06 01 04 ff ff 00 01.

        So , if I want the device-modbus send like :
    "00  01  00  00  00  06  00  04  00  00  00  02",
        How can I do?  Could you please give me an example to help me?

zhanglu0704, 2019-08-21:
And I can read the senser's values by C language application，and the sending  message like:
```
    char request[12];
    request[0]=0x00;
    request[1]=0x00;
    request[2]=0x00;
    request[3]=0x00;
    request[4]=0x00;
    request[5]=0x06;
    request[6]=0x00;
    request[7]=0x04;
    request[8]=0x00;
    request[9]=0x00;
    request[10]=0x00;
    request[11]=0x02; 
```
I can got values like:
```
----------------------------------
recv:13
69	01	00	00	00	07	00	04	04	01	2f	01	47	
--------------------------------
```

weichou1229, 2019-08-21:
Device-service always caches the data in the current version, if you update the device profile, please try `docker restart edge-device-modbus`.

zhanglu0704, 2019-08-21:
> Device-service always caches the data in the current version, if you update the device profile, please try `docker restart edge-device-modbus`.

 Thanks.
 I succeeded in my attempt.
 Previous modifications to profile did not take effect because there was no restart.
 In addition, I found that the startingAddress should be the same, and the number of
 registers depends on the number of device resources.
 So I Modified to
```
deviceResources:
    -
        name: "Temperature"
        description: " 0.1°C"
        attributes:
            { primaryTable: "INPUT_REGISTERS", startingAddress: "1" }
        properties:
            value:
                { type: "FLOAT32", readWrite: "R", scale: 0.1, floatEncoding: "Base64"}
            units:
                { type: "String", readWrite: "R", defaultValue: "°C"}
    -
        name: "Humidity"
        description: " %RH "
        attributes:
            { primaryTable: "INPUT_REGISTERS", startingAddress: "1" }
        properties:
            value:
                { type: "FLOAT32", readWrite: "R", scale: 0.1, floatEncoding: "Base64"}
            units:
                { type: "String", readWrite: "R", defaultValue: "%RH"}
```

closed 2019-08-21 by zhanglu0704