edgexfoundry/device-modbus-go #81: Enhancement:  suport ascii mode

bretagne-peiqi, 2019-09-23:
I am interested in adding ascii mode support for device-modbus-go. 

If it is helpful for our project. I am going to make a pull request. Thanks.

cloudxxx8, 2019-10-22:
@bretagne-peiqi is there any standard way to intepret ascii in Modbus?

bretagne-peiqi, 2019-10-27:
Hi, thanks for your reply. What do you mean by standard way to interpret ascii mode, interpret meas decode or somethings else as the links: http://www.simplymodbus.ca/ASCII.htmI saw you are using modbus library which implements tcp rtu&ascii modes.

for the driver part (device service), I have not seen different processes for these three modes.  

we are interested in this, waiting from your response :)

cloudxxx8, 2019-10-27:
@bretagne-peiqi Thanks, you are right.  It's in the standard.  Please feel free to open a PR in master branch, and we will help review the code.

cloudxxx8, 2025-12-17:
Thanks for the information, @sansmoraxz