edgexfoundry/device-modbus-go #88: Update the screenshot in "Modbus - Adding a Device to EdgeX" document

THEhEROLocker, 2019-10-18:
Link to Doc: https://docs.edgexfoundry.org/Ch-ExamplesAddingModbusDevice.html
Link to Screenshot: https://docs.edgexfoundry.org/_images/config_changes.png

I believe that this screenshot needs updating:

1) The configuration.toml needs to be in a 'docker' sub directory
2) The entrypoint flag "--registry" needs to be "--profile=docker"

weichou1229, 2019-10-23:
@THEhEROLocker Did you got any problem? I thought it's not necessary to use the profile flag.

THEhEROLocker, 2019-10-30:
It did not work for me when I tried it. But I only tried running it from the docker-compose as the document has it. 

When I try it exactly as the document has it, the modbus device service dies right away and I get this in the logs. 

```
Init: useRegistry: --confdir=/custom-config profile:  confDir: 
The port format of Registry path from argument is wrong:  strconv.Atoi: parsing "": invalid syntax
error loading config file: strconv.Atoi: parsing "": invalid syntax
```

weichou1229, 2019-10-30:
@THEhEROLocker
I may get your problem, the document is written for Edinburgh Release and the registry parameter changed.
We already create an issue to update the document https://github.com/edgexfoundry/edgex-docs/issues/32, I will update the document recently, thank you for your patience.

cloudxxx8, 2019-10-30:
@THEhEROLocker 
--registry needs the url right now,
for example, --registry=consul://edgex-core-consul:8500

weichou1229, 2019-12-10:
@THEhEROLocker We updated the documentation for fuji release.
https://github.com/edgexfoundry/edgex-docs/blob/fuji/examples/Ch-ExamplesAddingModbusDevice.rst
Could you take a look and then we can close this issue?

closed 2020-01-20 by THEhEROLocker