edgexfoundry/device-modbus-go #102: Modbus error Read & Write

aazkdj, 2019-11-28:
Hi!

I'm having some issues with the tutorial 6.3 that Edgex Foundry provides to add a device to edgex using ModbusPal.jar, a Modbus Device Simulator. In terms of connection, it works fine, but when asking for a get or put command via Postman, it always gives me an error like this:

**Get Command:** Handler - execReadCmd: error for Device: Power meter 1 cmd: Configuration, modbus: exception '1' (illegal function), function '131' /api/v1/device/f000172f-8249-4b63-81e3-626e9e0caa12/Configuration

**Put Command:** Handler - execWriteCmd: error for Device: Power meter 1 cmd: Configuration, handle write command request failed, err: modbus: exception '1' (illegal function), function '134' /api/v1/device/f000172f-8249-4b63-81e3-626e9e0caa12/Configuration

I have tried with another Modbus Device Simulator (just in case), which name is modslavesim3.06.jar , but nothing, the same issue. Could someone please help me?

Thanks in advance!

P.D: I leave here my Configuration.toml and the .yml profile. Also the pictures of the terminal showing the errors.

**ReadError**
[image]

**WriteError**
[image]

[modbus files.zip](https://github.com/edgexfoundry/device-modbus-go/files/3900451/modbus.files.zip)

weichou1229, 2019-12-03:
@aazkdj The exception can refer to http://www.simplymodbus.ca/exceptions.htm
> The function code received in the query is not an allowable action for the slave.  This may be because the function code is only applicable to newer devices, and was not implemented in the unit selected. **It could also indicate that the slave is in the wrong state to process a request of this type**, for example because it is unconfigured and is being asked to return register values. If a Poll Program Complete command was issued, this code indicates that no program function preceded it.

Please check...
Do you set up the HOLDING_REGISTERS register for device resources?
Do you click the "run" button to run the simulator?

aazkdj, 2019-12-03:
Hi @weichou1229 , thanks for your reply! Appreciate!

Yes, I did. I followed the Edgex Foundry tutorial and there tells me how to set up the holding registers and the coils of the slaves. I run the simulation and after that the docker-compose. I do not have any clue what is wrong on my project, I have check my scripts so many times... Hahaha

weichou1229, 2019-12-05:
@aazkdj Could you provide the full steps for reproducing the problem? 
The information include the OS, EdgeX version as shown in the [Bug Tracking guideline](https://wiki.edgexfoundry.org/display/FA/Contributor%27s+Guide#Contributor'sGuide-BugTracking)

aazkdj, 2019-12-09:
OS: Ubuntu 18.04.3 LTS
EdgeX version: Edinburgh

The steps
1. Install modbuspal (Modbus device simulator) and open it
2. Set the modbus devices (Create 2 slaves and add the holding registers and coil for each)
3. Add the value auto-generators as shown in the tutorial and bind value generator on the holding registers
4. Run the modbus device simulator
5. Set up the device profile (.yml) and configuration.toml before running docker-compose
6. Add device-modbus service to your docker-compose file
7. Start EdgeX Foundry on Docker by running the docker-compose
8. Execute commands (like get and put) to assure the proper functioning on the system
9. The shown error appears

weichou1229, 2019-12-10:
@aazkdj Could you try this [guideline](https://github.com/edgexfoundry/edgex-docs/blob/fuji/examples/Ch-ExamplesAddingModbusDevice.rst#run-amodbuspalmodbus-devicesimulator) to reinstall and set up the simulator?
After you reinstall the Modbuspal, you can load the[ xmpp file](https://github.com/edgexfoundry/edgex-docs/blob/fuji/examples/EdgeX_ExamplesModbus_SimulationPowerMeter.xmpp) directly.

aazkdj, 2019-12-10:
@weichou1229 now works properly! Thank you a lot!

closed 2019-12-10 by aazkdj