edgexfoundry/device-modbus-go #105: float32 reading render not as expect

212458312, 2019-12-04:
I have  worked with the Edgex Edinburgh version and configured a modbus device based on device profile as attached, everything works fine except the Energy float32 reading render not as expect.

When make a GET request against the following url:
http://192.168.56.10:48082/api/v1/device/name/Modbus-TCP-Device-0/command/Values

Get response as follows:
{
    "device": "Modbus-TCP-Device-0",
    "origin": 1575450182035170890,
    "readings": [
        {
            "origin": 1575450182010387787,
            "device": "Modbus-TCP-Device-0",
            "name": "**Energy**",
            "value": "**Q5ZDlg==**"
        },
        {
            "origin": 1575450182011151712,
            "device": "Modbus-TCP-Device-0",
            "name": "Power",
            "value": "100"
        },
        {
            "origin": 1575450182032879956,
            "device": "Modbus-TCP-Device-0",
            "name": "Voltage",
            "value": "140"
        }
    ]
}

Device-modbus service background logs as follows, it seems the modbus directive get the correct feedback from the slave device, then the reading is converted from  "300.528" to "Q5ZDlg==".

I can't figure out why this problem happens, please help me to confirm if it's a misconfiguration or a device-modbus issue, thanks.

edgex-device-modbus | level=INFO ts=2019-12-04T09:03:01.970429544Z app=edgex-device-modbus source=modbusclient.go:37 msg="Modbus client create TCP connection."
edgex-device-modbus | 2019/12/04 09:03:01 modbus: sending 00 01 00 00 00 06 01 03 0f a0 00 02
edgex-device-modbus | 2019/12/04 09:03:02 modbus: received 00 01 00 00 00 07 01 03 04 43 96 43 96
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.010363612Z app=edgex-device-modbus source=modbusclient.go:83 msg="Modbus client GetValue's results [67 150 67 150]"
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.01040143Z app=edgex-device-modbus source=deviceclient.go:139 msg="Transfer dataBytes to CommandValue(10, **300.528**) successful."
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.01041643Z app=edgex-device-modbus source=driver.go:144 msg="Read command finished. Cmd:Energy, Origin: 1575450182010387787, Float32: **Q5ZDlg==** \n"
edgex-device-modbus | 2019/12/04 09:03:02 modbus: sending 00 02 00 00 00 06 01 03 0f a2 00 01
edgex-device-modbus | 2019/12/04 09:03:02 modbus: received 00 02 00 00 00 05 01 03 02 00 64
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.011139849Z app=edgex-device-modbus source=modbusclient.go:83 msg="Modbus client GetValue's results [0 100]"
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.011156087Z app=edgex-device-modbus source=deviceclient.go:139 msg="Transfer dataBytes to CommandValue(3, 100) successful."
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.011168787Z app=edgex-device-modbus source=driver.go:144 msg="Read command finished. Cmd:Power, Origin: 1575450182011151712, Uint16: 100 \n"
edgex-device-modbus | 2019/12/04 09:03:02 modbus: sending 00 03 00 00 00 06 01 03 0f b0 00 01
edgex-device-modbus | 2019/12/04 09:03:02 modbus: received 00 03 00 00 00 05 01 03 02 00 8c
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.030948216Z app=edgex-device-modbus source=modbusclient.go:83 msg="Modbus client GetValue's results [0 140]"
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.032960758Z app=edgex-device-modbus source=deviceclient.go:139 msg="Transfer dataBytes to CommandValue(3, 140) successful."
edgex-device-modbus | level=INFO ts=2019-12-04T09:03:02.032980858Z app=edgex-device-modbus source=driver.go:144 msg="Read command finished. Cmd:Voltage, Origin: 1575450182032879956, Uint16: 140 \n"

212458312, 2019-12-04:
[GS1-10P5.profile.yaml.txt.txt](https://github.com/edgexfoundry/device-modbus-go/files/3921812/GS1-10P5.profile.yaml.txt.txt)

212458312, 2019-12-04:
[GS1-10P5.profile.yaml.txt](https://github.com/edgexfoundry/device-modbus-go/files/3921816/GS1-10P5.profile.yaml.txt)

cloudxxx8, 2019-12-04:
@212458312 the reading value is as expected.  we use base64 encoding for floating value by default.  you need to decode the string value or in the profile, set the **floatEncoding** as **eNotation**

For example,
https://github.com/edgexfoundry/device-modbus-go/blob/fa727a71fa3d4c695f6f29eeafb4dd7c8c2f2674/cmd/res/example/modbus.test.device.profile.yml#L72

212458312, 2019-12-05:
@cloudxxx8 thank you, I have verified, it works!

closed 2019-12-05 by 212458312