edgexfoundry/device-modbus-go #147: we find that Send modbus message if Real device no response modbus message ,Then edgex-device-mobus Not sending message

chenkewangqq, 2020-04-28:
How is this configured

cloudxxx8, 2020-04-28:
please write down the clear reproduce steps and logs.  thanks

chenkewangqq, 2020-04-28:
> please write down the clear reproduce steps and logs. thanks

2020/04/28 08:40:56 modbus: received 31 0a 31 0a
level=INFO ts=2020-04-28T08:40:56.036811604Z app=edgex-device-modbus source=driver.go:116 msg="Read command failed. Cmd:J-1 err:modbus: response slave id '49' does not match request '3' \n"
level=ERROR ts=2020-04-28T08:40:56.036872436Z app=edgex-device-modbus source=executor.go:46 msg="AutoEvent - error occurs when reading resource YxCommand"
level=INFO ts=2020-04-28T08:40:56.037756097Z app=edgex-device-modbus source=modbusclient.go:41 msg="Modbus client create RTU connection."
2020/04/28 08:40:56 modbus: sending 15 03 00 25 00 01 96 d5
level=ERROR ts=2020-04-28T08:40:56.549042488Z app=edgex-device-modbus source=command.go:72 msg="YcCommand for Device: 16k not found; get"
level=ERROR ts=2020-04-28T08:40:56.549174071Z app=edgex-device-modbus source=executor.go:46 msg="AutoEvent - error occurs when reading resource YcCommand"

chenkewangqq, 2020-04-28:
> please write down the clear reproduce steps and logs. thanks

If the device does not respond to modbus-rtu, the rtu command stops

chenkewangqq, 2020-04-28:
> please write down the clear reproduce steps and logs. thanks

I need to know how to configure the rtu message timeout

chenkewangqq, 2020-04-28:
> please write down the clear reproduce steps and logs. thanks

Can you speak Chinese? Do you have qq or WeChat

weichou1229, 2020-04-28:
> level=INFO ts=2020-04-28T08:40:56.036811604Z app=edgex-device-modbus source=driver.go:116 msg="Read command failed. Cmd:J-1 err:modbus: response slave id '49' does not match request '3' \n"

> level=ERROR ts=2020-04-28T08:40:56.549042488Z app=edgex-device-modbus source=command.go:72 msg="YcCommand for Device: 16k not found; get"

It seems you set up the wrong slave id and use the wrong command name to execute read command.

chenkewangqq, 2020-04-28:
> > level = INFO ts = 2020-04-28T08：40：56.036811604Z app = edgex-device-modbus source = driver.go：116 msg =“读取命令失败。Cmd：J-1 err：modbus：响应从站ID '49 '与请求'3'不匹配\ n“
> 
> > level = ERROR ts = 2020-04-28T08：40：56.549042488Z app = edgex-device-modbus source = command.go：72 msg =“设备的YcCommand：未找到16k；获取”
> 
> 似乎您设置了错误的从站ID，并使用了错误的命令名称来执行读取命令。

no, I don't have a real device at all,What I'm trying to say is how do you get orders executed,how to The rtu command continues

cloudxxx8, 2020-04-28:
what do you mean the rtu command continues?  I don't really understand your question.

chenkewangqq, 2020-04-28:
> 您是什么意思rtu命令继续？我不太了解您的问题。

how to set  this  , client.RTUClientHandler.IdleTimeout = 0 ,if i set client.RTUClientHandler.IdleTimeout =1 *time.second ,then real device no response,then timeout?

cloudxxx8, 2020-04-29:
the timeout is not configurable currently.  please reach me on slack if you would like to chat in Chinese

chenkewangqq, 2020-05-01:
> 超时当前无法配置。如果您想用中文聊天，请悠闲地与我联系
可以通过docker配置多个edgex-device-modbus程序吗，互相如何不冲突进行采集

chenkewangqq, 2020-05-01:
> the timeout is not configurable currently. please reach me on slack if you would like to chat in Chinese
可以通过docker配置多个edgex-device-modbus程序吗，互相如何不冲突进行采集

chenkewangqq, 2020-05-01:
> 超时当前无法配置。如果您想用中文聊天，请悠闲地与我联系

Where do I want to see the source code for device-profile

weichou1229, 2020-05-04:
The device profile definition:
https://github.com/edgexfoundry/go-mod-core-contracts/blob/master/models/deviceprofile.go

You can also ask the question on the slack https://slack.com/

weichou1229, 2020-08-04:
Close issue. Will add the timeout parameter when creating the Modbus connection at the issue https://github.com/edgexfoundry/device-modbus-go/issues/108

closed 2020-08-04 by weichou1229