edgexfoundry/device-modbus-go #156: Questions about github.com/goburrow/modbus go import

tonyespy, 2020-06-23:
While working on the PRs to update all of our device services to the latest device-sdk-go for our Geneva maintenance release, I noticed that the underlying modbus implementation used by device-modbus-go had a version tag of v0.1.0.

Looking at the actual package [github.com/goburrow/modbus](https://github.com/goburrow/modbus):

 * it has one release from 2018
 * the project has four contributors
 * the last commit was two years ago
 * there are 22 open issues
 * it has 8 open PRs, one actually opened 23 days ago

To me, this looks like a dead project, or maybe just one on life support, and thus question it's usage as the basis for this device service...

**Note** - this was originally reported in edgex-go issue [#1947](https://github.com/edgexfoundry/edgex-go/issues/1947) (Develop process for security vetting of 3rd party components).

iain-anderson, 2020-07-13:
Discussed in WG Jun-29-2020, this appears to be the best/only option currently available. If issues arise we will have to determine whether it is possible to get an upstream fix or fork the library ourselves

jpwhitemn, 2021-03-08:
@cloudxxx8 - when looking at moving this Modbus device service to v2 APIs with new V2 SDK, can you look to see how easy it is to move to a new modbus library.  @tonyespy concerned that this library is no longer being maintained.

jpwhitemn, 2021-03-08:
@cloudxxx8 - if the library can't be replaced, how hard would this to be fork and maintain ourselves (per the DS WG meeting of 3/8/21)

cloudxxx8, 2021-03-10:
@jpwhitemn @tonyespy I made some research, and didn't find other better golang open source Modbus driver.
If we want to fork and maintain on our own, it's achievable, but we need to find an owner.  It's hard to add more new feature, but should be able to fix bugs.  We can fork it until we encounter any real bug.

cloudxxx8, 2023-12-12:
decided in the Device WG 11-DEC-2023, we won't replace the lib until users encounter real issues, and we can fork and resolve it on ourown.
closing this issue.

potential replacement: https://github.com/simonvetter/modbus

closed 2023-12-12 by cloudxxx8