edgexfoundry/device-modbus-go #167: Modbus: exception '1' (illegal function) in read & write operation

flucciarini, 2020-08-25:
Hi everyone, 
I've follow this guide [https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice/](https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice/) to simulate a modbus device and I follow the step at Set Up After Starting Services but when I try to execute a comman I encounter an error (get or set, both has a problem).
The problem is the follow: 
**_Handler - execReadCmd: error for Device: Modbus-TCP-Device-2 cmd: Configuration, modbus: exception '1' (illegal function), function '131' /api/v1/device/9970d0be-ea6c-425e-9553-921be3a804f1/Configuration._** 
The connection is ok. I've checked MudbusPal configuration a lot of time and seems ok. 
Also in device-modbus container I found this error for scheduled event 
**_edgex-device-modbus | level=INFO ts=2020-08-25T14:26:49.083715394Z app=edgex-device-modbus source=modbusclient.go:37 msg="Modbus client create TCP connection."
edgex-device-modbus | 2020/08/25 14:26:49 modbus: sending 00 01 00 00 00 06 02 03 0f a0 00 02
edgex-device-modbus | 2020/08/25 14:26:49 modbus: received 00 01 00 00 00 03 02 83 01
edgex-device-modbus | level=INFO ts=2020-08-25T14:26:49.084837804Z app=edgex-device-modbus source=driver.go:121 msg="Read command failed. Cmd:Energy err:modbus: exception '1' (illegal function), function '131' \n"
edgex-device-modbus | level=ERROR ts=2020-08-25T14:26:49.084909114Z app=edgex-device-modbus source=executor.go:52 msg="AutoEvent - error occurs when reading resource Values"_**

OS: CentOS 7
EdgeX version: Geneva
ModbusPal: 1.6b
device-modbus-go: 1.2.1

The steps

Install modbuspal (Modbus device simulator) and open it
Set the modbus devices (Create 2 slaves and add the holding registers and coil for each)
Add the value auto-generators as shown in the tutorial and bind value generator on the holding registers
Run the modbus device simulator
Add device-modbus service to your docker-compose file
Start EdgeX Foundry on Docker by running the docker-compose
Post DeviceProfile
Check DeviceProfile with GET /api/v1/deviceProfile (OK)
Post Device
Check Device with GET /api/v1/device (OK)
Execute commands (like get and put) to assure the proper functioning on the system
The shown error appears
Thanks in advance!

weichou1229, 2020-08-26:
@flucciarini Didi you click the ModbusPal's Run button?

weichou1229, 2020-08-26:
@flucciarini You may read the invalid address, you can check the ModbusPal's addresses are setup correctly.

The exception code refer to http://www.simplymodbus.ca/exceptions.htm
A value contained in the query data field **is not an allowable value for the slave**.  This indicates a fault in the structure of remainder of a complex request, **such as that the implied length is incorrect**. It specifically does NOT mean that a data item submitted for storage in a register has a value outside the expectation of the application program, since the MODBUS protocol is unaware of the significance of any particular value of any particular register.

flucciarini, 2020-08-26:
@weichou1229 I setup up ModbusPal's address, deviceProfile and device configuration as shown in this guide [https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice](https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice) except obviously the various ip address

I noticed that the request sended to modbus device is 
modbus: sending 00 01 00 00 00 06 01 03 0f a0 00 02 and I think it's correct (refer in [http://www.simplymodbus.ca/TCP.htm](http://www.simplymodbus.ca/TCP.htm))
0001: Transaction Identifier
0000: Protocol Identifier
0006: Message Length (6 bytes to follow)
01: The Unit Identifier  (slave n. 1)
03: The Function Code (read Analog Output Holding Registers)
0fa0: The Data Address of the first register requested. (44001-40001 = 4000 =FA0 hex)
0002: The total number of registers requested. (read 2 registers 44001 to 440002)

ModbusPal configuration in attacched png (both slave had equals configuration)
[image]

weichou1229, 2020-08-28:
@flucciarini 
I can't recreate your issue. 

Did you setup the ModbusPal by loading the xmpp file? 
It might cause same exception, the issue can refer to https://sourceforge.net/p/modbuspal/discussion/899955/thread/72cf35ee/

wlgardner, 2020-08-29:
Would this be an issue because of the default -1 on the startAddress in deviceClient?

weichou1229, 2020-08-30:
No, 
4001 is starting address(Register number)
4000 is data address
https://www.simplymodbus.ca/FAQ.htm#Stored

flucciarini, 2020-08-31:
@weichou1229 
I reset the ModbusPal and reconfigure it and all work fine. The problem occurs only when load xmpp file.
The fixed version not work for me :( but there is no problem I reconfigure the simulator when need

Thanks

weichou1229, 2020-08-31:
@flucciarini 
> I reset the ModbusPal and reconfigure it and all work fine.

Great!

> The fixed version not work for me :

I've tested on the Ubuntu18 and Ubuntu20 and it's work for me. Or you can try other free Modbus simulator like https://www.modbusdriver.com/diagslave.html

Since it's not device-modbus's problem, may I close this issue?

flucciarini, 2020-08-31:
@weichou1229 
Yes it can close for me

closed 2020-08-31 by cloudxxx8