edgexfoundry/device-modbus-go #168: feat: Optimize the Modbus client's error log

weichou1229, 2020-08-26:
User may setup the Modbus device incorrectly and device-modbus throw error log like the following message:
> Handler - execReadCmd: error for Device: Modbus TCP test device cmd: HVACValues, modbus: exception '2' (illegal data address), function '131' /api/v1/device/f6d51c16-1a47-4020-a579-091e56723124/HVACValues 

So it worth to improve the error log according to online docs:
https://en.wikipedia.org/wiki/Modbus#Exception_responses
http://www.simplymodbus.ca/exceptions.htm

vijay-hcl-epl, 2020-12-18:
Hi @weichou1229,

I am able to recreate the issue. This particular error log comes from device-sdk-go in the file command.go I also went through the issue# 167.

As per https://www.simplymodbus.ca/exceptions.htm, for "Illegal data address" (as mentioned in the issue), the exception code is 02 and the exception description is shown below

> The data address received in the query is not an allowable address for the slave. More specifically, the combination of reference number and transfer length is invalid. For a controller with 100 registers, a request with offset 96 and length 4 would succeed, a request with offset 96 and length 5 will generate exception 

Could you please let us know what exactly needs to be optimized in this error log?

Thank you,
Vijay

weichou1229, 2020-12-18:
No problem, it need to parse the`err` and return more properly error message.
```
	if err != nil {
		return err
	}
```
https://github.com/edgexfoundry/device-modbus-go/blob/master/internal/driver/modbusclient.go#L80
https://github.com/edgexfoundry/device-modbus-go/blob/master/internal/driver/modbusclient.go#L115

vijay-hcl-epl, 2020-12-25:
Thank you @weichou1229  for the suggestion.

The below error log comes from 3 files

> Handler - execReadCmd: error for Device: Modbus TCP test device cmd: HVACValues, modbus: exception '2' (illegal data address), function '131' /api/v1/device/f6d51c16-1a47-4020-a579-091e56723124/HVACValues

The files are restfuncs.go, command.go and modbusclient.go.

Command.go prints below text 

> Handler - execReadCmd: error for Device: Modbus TCP test device cmd: HVACValues

restfuncs.go prints below text

> /api/v1/device/f6d51c16-1a47-4020-a579-091e56723124/HVACValues

modbusclient.go prints below text

> modbus: exception '2' (illegal data address), function '131'

I understand we need to change only modbusclient.go file. However, I feel all the information in the above text(from modbusclient.go) are needed to explain the modbus exception. Could you please be specific about how exactly the above text needs to be optimized? 

Shall I parse and change it to something like below? Please suggest

> Exception Code: 131 (illegal data address)

Thank You,
Vijay

cloudxxx8, 2021-01-09:
@vijay-hcl-epl please feel free to open the PR to improve the message according to your experience.  We haven't had good suggestion so far.  Let's discuss in the PR.
We didn't notice the error message of Golang convention is starting with lower case, so please also help correct this part.
Thanks.

vijay-hcl-epl, 2021-01-10:
@cloudxxx8 , Thank you for the response. 

When analysed more, found that the below error comes from the file https://github.com/goburrow/modbus/blob/606c02f4eef527a1d4cf7d8733d5fd7ba34f91d8/modbus.go#L74

> modbus: exception '2' (illegal data address), function '131'

This statement is in the package "github.com/goburrow/modbus".  

The error logs are coming from three different files - restfuncs.go, command.go & modbus.go. modbusclient.go just receives and prints it. I will check how it can be optimised and come back.

Thank you,
Vijay

vijay-hcl-epl, 2021-01-11:
@cloudxxx8, I tried to make changes locally inside modbusclient.go and submit a PR but not able to do so as the logs are coming from 3 different projects - net/http, device-sdk-go and goburrow/modbus. The logs coming from goburrow/modbus starts with lower case. 

The files which prints the logs have been listed below

https://github.com/edgexfoundry/device-sdk-go/blob/909cc88fb1ad949568bf33336c87c0ea6bd2f5d5/internal/handler/command.go#L304
https://github.com/edgexfoundry/device-sdk-go/blob/909cc88fb1ad949568bf33336c87c0ea6bd2f5d5/internal/controller/restfuncs.go#L129
https://github.com/goburrow/modbus/blob/606c02f4eef527a1d4cf7d8733d5fd7ba34f91d8/modbus.go#L74
https://github.com/golang/go/blob/520f3b72db7befab2028d9a47376267cf2d274a9/src/net/http/server.go#L2078

This log is printed as a http REST response to the user request. modbusclient.go receives the logs from the other packages and forwards to net/http package. So making the changes locally to modbusclient.go is not sufficient to solve this issue.

cloudxxx8, 2021-01-11:
@vijay-hcl-epl we are going to make some big refactors on device-sdk-go for v2, and you can feel free to open some PRs on this repository to improve the log message.
however, we might not able to modify anything on the Modbus library, but we can wrap or convert to some better messages in device-modbus-go.
Please just do the best what we can do.  Thank you.