edgexfoundry/device-modbus-go #178: How to connect Modbus device to Edgex Foundry?

trantram310, 2020-10-12:
Hi, I'm newbie in Edgex Foundry. I want to connect Modbus device( inverter Delta-vfd-l) to Device Service. I install Edgex Foundry on rasperry Pi 3 through **sudo snap install edgexfoundry** and Edgex UI showed at **http://((your-raspberry-pir-url):4000/**. I read lots of documents but all guide use Docker to deploy Edgex. How to connect Modbus device to Device Service not use Docker? Please help me

cloudxxx8, 2020-10-12:
@tonyespy @anonymouse64 it's question about using snap.  could you please help answer him?  thanks.

trantram310, 2020-10-13:
@tonyespy @anonymouse64  Please help me with EgdeX.

trantram310, 2020-10-13:
@cloudxxx8  In case I use Docker and Docker-compose to connect real modbus device (Inverter) to EdgeX Foundry? Can you give me a document or link about this stuff, please? I use Rasperry Pi 3 and I installed  Ubuntu 18.04.5 server on SD card.

cloudxxx8, 2020-10-13:
Here is a tutorial
https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice/

trantram310, 2020-10-15:
@cloudxxx8  I updated device profile but at Device Service not show edgex-device-modbus. what's wrong?
[image]

cloudxxx8, 2020-10-15:
I am not sure whether it is an UI issue.  Could you please try to call the REST API to see your Device Service?
http://host:48081/api/v1/deviceservice

https://app.swaggerhub.com/apis-docs/EdgeXFoundry1/core-metadata/1.2.0#/default/get_v1_deviceservice

trantram310, 2020-10-19:
@cloudxxx8 I check device service then I know Modbus device service is Enable. But how can I check device connection in metadata and registered device in core data? Please help me !
[image]

trantram310, 2020-10-19:
Ah,I check device through **curl -s http://<Rasperry-IP>:48081/api/v1/device | jq** then I know device modbus is not exist. So, maybe device is not registered in metadata. How can I check this?

trantram310, 2020-10-19:
@cloudxxx8  I can use this comment????
[image]

cloudxxx8, 2020-10-19:
@trantram310 yes, please use POST API to create the Device.  You have to create Device Profile first.
GUI service also calls this API to create a Device.

trantram310, 2020-10-19:
@cloudxxx8  Ah, I updated Modbus Device profile successfully, but when I use POST API to create device that it's error ( **error: invalid character 'A' looking for beginning of object key string**)
I'm confused about Configuration.toml file cuz it dont explain about protocol.
[image]
[image]

cloudxxx8, 2020-10-19:
Configuration is for the service itself, and you should follow the JSON schema defined in the API to create the device
https://app.swaggerhub.com/apis-docs/EdgeXFoundry1/core-metadata/1.2.0#/default/post_v1_device

The Modbus example shows in the "Set Up After Starting Services" section step 3.
https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice/

trantram310, 2020-10-20:
@cloudxxx8  I read document you give and I use POST API with statement. But it not create Modbus device for me. I check device profile in metadata then metadata has device profile.
[image]

trantram310, 2020-10-20:
@cloudxxx8  My code and error
[image]

trantram310, 2020-10-20:
This is my Modbus device profile.
[image]

cloudxxx8, 2020-10-20:
it is a JSON format issue.  please remove the comma in the last line of each {} block
for example,
"unitID":"1"**,**

and 

"profile":{"name":"Test-Device-Modbus"}**,**

trantram310, 2020-10-20:
@cloudxxx8  Maybe I miss DeviceID. So I fix 2 errors as you. It show **Device ID and Name are both blank**

cloudxxx8, 2020-10-20:
are you calling PUT or POST?  You don't need to assign Id for POST

trantram310, 2020-10-20:
@cloudxxx8 I call POST api.
[image]

cloudxxx8, 2020-10-20:
you have an additional space in the profile section

trantram310, 2020-10-20:
@cloudxxx8  So, How can I check this error? Maybe in my device profile I put to metadata?

trantram310, 2020-10-20:
This is my device profile.
[image]

trantram310, 2020-10-20:
@cloudxxx8  I check addressable then: 
[image]

cloudxxx8, 2020-10-20:
No, the addressable is for Device Service, not Device.
Your profile is fine.

The error message **Device ID and Name are both blank** is incorrect.  It should say: **Device Profile ID and Name are both blank**

The problem is caused by your profile section in your Device creation:

"profile":{"name ": "Test-Device-Modbus"}

There is an additional space in **"name "**,  so it says there is no name.

trantram310, 2020-10-20:
@cloudxxx8  Yes, It returns me a ID. How can I get value from my Modbus device sensor. I use **curl 192.168.1.22:48082/api/v1/device/{deviceId}/Lux** , but it returns me  **404 page not found**

cloudxxx8, 2020-10-20:
if you use device id, you should also use command id.
instead, you can use device name and command name API
https://app.swaggerhub.com/apis-docs/EdgeXFoundry1/core-command/1.2.0#/default/get_v1_device_name__name__command__commandname_

trantram310, 2020-10-20:
I check device command use **curl 192.168.1.22:48082/api/v1/device | json_pp** then know. How can i get sensor data from this?
[image]

trantram310, 2020-10-20:
@cloudxxx8 I check device through  **curl 192.168.1.22:48081/api/v1/device | jq** the know
[image]

trantram310, 2020-10-20:
I check 192.168.1.22:48082/api/v1/device to get command then Modbus not get url.
[image]

cloudxxx8, 2020-10-20:
please paste the full result of 192.168.1.22:48082/api/v1/device in text

trantram310, 2020-10-20:
[Modbus..txt](https://github.com/edgexfoundry/device-modbus-go/files/5409033/Modbus.txt)
It's little visionable !!!

cloudxxx8, 2020-10-20:
there are some URL fields in the result, and you should use them
Or, it is clear in the **Execute GET command** section of the document
192.168.1.22:48082/api/v1/device/name/Test-Device-Modbus/command/Lux

It is the API I mentioned

trantram310, 2020-10-20:
@cloudxxx8 wait!  I check **192.168.1.22:48082/api/v1/device** and see it not show commands which I code in Modbus device profile. Is it real problems? This is all of about Modbus device when I check **192.168.1.22:48082/api/v1/device** again.
[image]

trantram310, 2020-10-20:
@cloudxxx8 When I use **curl 192.168.1.22:48082/api/v1/device/6a29dc9b-c307-4f95-9b5e-4fed431ff6c9/Lux** then it return me **404 page not found**. it not have a lux value

cloudxxx8, 2020-10-21:
again, you hit thewrong API. 
`curl 192.168.1.22:48082/api/v1/device/6a29dc9b-c307-4f95-9b5e-4fed431ff6c9/Lux` is wrong

Please see the API document.  The API is not api/v1/device/{device id}/{command name}
it is 
​/v1​/device​/{id}​/command​/{commandid}
or
/v1/device/name/{name}/command/{commandname}

I also notice that the command sections are wrong in your profile.

Please see the `Set Up Device Profile` section. You need to define `deviceCommands` and `coreCommands`
https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice/

However, your profile contains `resources` and `commands`.  Thus, there is no command in your Device can be called.

trantram310, 2020-10-21:
@cloudxxx8 Yes, I do as you said about modbus device profile, I fix it and when I call **curl 192.168.1.22:48082/api/v1/device** then it return me commands. But when I call statements as command then it dont return me sensor data as I defined in device profile. It return me:
[image]
[image]

trantram310, 2020-10-21:
@cloudxxx8 I check this commands and return me this errors.
[image]

cloudxxx8, 2020-10-21:
it's a permission issue.  you should:

1. chmod /dev/ttyUSB0 to give the proper permission
2. add volume map in the docker-compose file, and let the container can access it
https://docs.docker.com/compose/compose-file/#devices

trantram310, 2020-10-21:
@cloudxxx8 I'm deploying through Snap store, not docker-compose file. So, I can't know the second you said.

trantram310, 2020-10-21:
@cloudxxx8 I fix and check again. It returns errors:
[image]

cloudxxx8, 2020-10-21:
You have connected to the Device with the wrong starting address.
Please confirm the deviceResources - attributes section is correct and match the Modbus Device.
Also, the data type should be correct
The starting address is 1 base, not 0 base.

For example from the document:
```
deviceResources:
  -
    name: "Current"
    description: "Average current of all phases"
    attributes:
      { primaryTable: "HOLDING_REGISTERS", startingAddress: "6" }
    properties:
      value:
        { type: "UINT16", scale: "1"}
      units:
        { type: "String", readWrite: "R", defaultValue: "min"}
```

trantram310, 2020-10-21:
@cloudxxx8 I fix as you said. But it return me old error. This is my device document and deviceResources I defined.
[image]
[image]
And here is my Post API to create new device.
[image]

I dont know where is problem I meet ?

trantram310, 2020-10-21:
@cloudxxx8  This is my full Device profile. Please help me find problem, I really need your help.
[image]

cloudxxx8, 2020-10-21:
The starting address is 1 base, so you can try
`{ primaryTable: "HOLDING_REGISTERS", startingAddress: "7" }`

trantram310, 2020-10-21:
@cloudxxx8  It returns me old errors. But i dont know when I call this command then it return is that Device not found while I created it??
[image]

trantram310, 2020-10-22:
@cloudxxx8 When I check port UART I use to read sensor data then I see. 
[image]
I follow steps in document but it return me error.I dont know where is error?

trantram310, 2020-10-22:
@cloudxxx8 I dont know here is my problem. I check file configuration.toml in /snap/edgex-device-modbus/1256/config/device-modbus/res then I see it not define [Devicelist]. Here is all in file configuration.toml in edgex-device-modbus packet I install through Snap
[image]
[image]

cloudxxx8, 2020-10-22:
please restart the modbus device service and try again

tonyespy, 2020-10-22:
A couple of comments...

 * Please use this issue for further discussion of your problem, as it's confusing when the discussion is happening in more than one place (i.e. this issue & slack).
 * Please confirm that your trying to use the device service in RTU mode (i.e. a serial device). If so, what is the actual hardware (i.e. not the Linux serial device) you're working with?
 * In an earlier comment, you were trying to add a device with `Address=/dev/ttyUSB0`. As the device-modbus snap is fully confined, AppArmor will prevent access to hardware interfaces by default. In fact, in the thread on #help, you posted output from `dmesg` which shows AppArmor denying access to the device service when it tries to open `/dev/ttyUSB0`. We haven't yet added support for the snap to securely access serial devices, so in order to use the snap for RTU Modbus, you need to install the snap using `--devmode` option. I'll work with my team to make sure this is documented. This is similar to what you'd need to do if you ran the device service via docker (i.e. you'd need to specify the container as privileged). I'll also add an issue to the device service to add support for secure access to serial devices for RTU mode.

lingdujunshang, 2022-06-24:
> @cloudxxx8 我更新了设备配置文件，但在设备服务中没有显示 edgex-device-modbus。怎么了？ [image]
hello，i am develop neither docker nor snap, i have edgex-go and edgex-ui-go ready.but i just can not login ui page,i do not know how to get a token to login ..do u have any idea pls?

cloudxxx8, 2022-06-24:
please check the log of device-modbus container

yoiberdev, 2024-11-29:
Me podrian ayudar a conectar mi dispositivo modbus físico?
Apesar que lo configure con el pureto correcto me vota code: 500 , message: request failed, status code: 500, err: {"apiVersion":"v3","message":"error reading DeviceCommand DO for Modbus-RTU-IO-Module -\u003e dial tcp: lookup /dev/ttyUSB0: no such host","statusCode":500}

yoiber@yoiber:~/Documentos/edgex/edgex-compose/compose-builder$ sudo dmesg | grep tty
[    0.061972] printk: legacy console [tty0] enabled
[    0.327186] 00:01: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    0.350951] 00:02: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
[    0.374702] 00:04: ttyS4 at I/O 0x240 (irq = 11, base_baud = 115200) is a 16550A
[    0.399381] 00:05: ttyS5 at I/O 0x248 (irq = 11, base_baud = 115200) is a 16550A
[    2.620808] usb 1-1: pl2303 converter now attached to ttyUSB0
[ 4302.556426] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[ 4386.601082] usb 1-7.2: pl2303 converter now attached to ttyUSB0
[ 4478.259331] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
yoiber@yoiber:~/Documentos/edgex/edgex-compose/compose-builder$

[image]

cloudxxx8, 2024-12-02:
close this issue as no activity

closed 2024-12-02 by cloudxxx8