edgexfoundry/device-modbus-go #187: Modbus RTU connection error

ThomasGeor, 2020-11-09:
I am trying to connect a Modbus RTU slave device through UART and on address /dev/ttyUSB0 and i am getting the following error when i am trying to send a get command : 

Postman LOGS :
**Handler - execReadCmd: error for Device: em340_rtu_slave_modbus cmd: MeterReadings, no such file or directory /api/v1/device/2296404e-b11a-4e14-81e0-5730695c39b8/MeterReadings**

device-modbus LOGS:
**level=INFO ts=2020-11-09T16:08:21.589695726Z app=edgex-device-modbus source=modbusclient.go:41 msg="Modbus client create RTU connection."
level=INFO ts=2020-11-09T16:08:21.589754748Z app=edgex-device-modbus source=driver.go:111 msg="Read command OpenConnection failed. err:no such file or directory \n"**

I have already made sure that /dev/ttyUSB0 is connected correctly and displayed when i am using ** ls /dev/tty* **.
I have also made sure that baudrates and serial configurations are the same for my PC and the targetted device. 
When using putty to test a UART echo, it is done correctly. The problem is that edgeX can't open the connection as shown above. Is there any solution to that? What does this error message mean except from the fact that the serial connection cannot be opened?

weichou1229, 2020-11-10:
@ThomasGeor Did you try to mapping the host /dev/ttyUSB0 to your docker container?
For this case, we opened an issue to improve the doc.

cloudxxx8, 2020-11-10:
Please use 
https://docs.docker.com/engine/reference/commandline/run/#add-host-device-to-container---device
or
https://docs.docker.com/compose/compose-file/#devices
to map your device path

ThomasGeor, 2020-11-10:
Thanks for the help! I solved it by adding the following to my docker-compose file : 

[image]

closed 2020-11-10 by ThomasGeor