edgexfoundry/device-modbus-go #202: Communication between modbus and MQTT from a PLC o VFD

CesarCastillo-90, 2020-12-14:
Hello everyone, I hope you are well. I have managed to successfully integrate the ModbusPal emulator with EdgeX Foundry's modbus microservice. However, I have doubts about integrating any device with and seeing the results through MQTT. On the other hand, data management has been carried out from the southern border to the northern border, but I have not been able to communicate in the opposite direction. If anyone knows how to do this, I will be grateful. I look forward to your valuable comments, have a great day.

CesarCastillo-90, 2020-12-14:
ok @cloudxxx8, How can I communicate a PLC by MODBUS protocol to MQTT through EdgeX Foundry and in turn to write data from MQTT to PLC? Thank you so much!

cloudxxx8, 2020-12-15:
You could set up two App Services, one for exporting PLC data to MQTT broker, and another receiving MQTT message and send http command to Core Command service.

CesarCastillo-90, 2020-12-15:
Hi mr. @cloudxxx8, could you guide me in this?, I'm using docker compose hanoi
[docker-compose-hanoi-arm64.zip](https://github.com/edgexfoundry/device-modbus-go/files/5696459/docker-compose-hanoi-arm64.zip)

weichou1229, 2020-12-16:
If you want to push data to northbound's MQTT broker, you can try something like AppServiceConfigurable's `mqttsend`
https://docs.edgexfoundry.org/1.3/microservices/application/AppServiceConfigurable/#mqttsend

CesarCastillo-90, 2020-12-16:
Hi mr. @weichou1229, I have communication from the south border to the north via MQTT, however I do not have the opposite direction. If you can guide me in this regard, I would be grateful.

weichou1229, 2020-12-17:
You can execute read/write command via core-command service,
see example https://docs.edgexfoundry.org/1.2/examples/Ch-ExamplesAddingModbusDevice/#execute-commands

CesarCastillo-90, 2020-12-17:
Hi mr. @weichou1229,Yes, I was able to use the execution commands and see that I could read and write results in the ModbusPal emulator, even see the data in the MQTT broker. However, I don't know how to write data as a publisher to the MQTT broker to see it reflected in the ModbusPal emulator, or well, I don't know if that's possible. I appreciate your guidance.

CesarCastillo-90, 2020-12-17:
ok mr. @weichou1229, I go to see it, thank you so much!

weichou1229, 2020-12-17:
>  I don't know how to write data as a publisher to the MQTT broker to see it reflected in the ModbusPal emulator

Currently, App-Service-Configurable doesn't provide such feature, but you can create one for yourself. The introduction and example link can refer to https://github.com/edgexfoundry/app-functions-sdk-go/blob/master/README.md

CesarCastillo-90, 2020-12-17:
ok mr. @weichou1229 thank you for your help.

CesarCastillo-90, 2020-12-17:
Understood, I will do tests, if I have any concerns I will consult you again, thank you very much mr. @weichou1229.

karima-se, 2022-04-11:
Hello @weichou1229 , would you like guid me to send data obtaned from ModbusPal simulator to MQTT broker please, I am using Irland version(v 2.0),  I would be grateful.

weichou1229, 2022-05-05:
The compose-builder can setup a app service to export events to MQTT broker
https://github.com/edgexfoundry/edgex-compose/tree/main/compose-builder

> add-asc-mqtt-export.yml Application Service Configurable extending compose file, which adds the App Service MQTT Export service