edgexfoundry/device-modbus-go #203: Regarding MODBUS communication through an MQTT broker.

CesarCastillo-90, 2020-12-14:
Hello everyone, I hope you are well. I have the following case: This code is the MODBUS response that I receive through the MQTT topic.

**{"id":"26cd17d9-aa6d-45d7-b3dc-1fd5fadd5ca1","device":"Modbus-TCP-Device-2","created":1607981337586,"origin":1607981337581736137,"readings":[{"id":"77437133-9d56-4b5a-8445-ca19cd39b902","origin":1607981337473057323,"device":"Modbus-TCP-Device-2","name":"DemandWindowSize","value":"500","valueType":"Uint16"},{"id":"2345f849-0f64-4ada-9ba6-cf3b40ad6710","origin":1607981337581112262,"device":"Modbus-TCP-Device-2","name":"LineFrequency","value":"120","valueType":"Uint16"}]}**

However, I don't know how to post to the topic to upload data to the ModbusPal simulator. I appreciate the guidance you can give me, I look forward to your valuable comments.

CesarCastillo-90, 2020-12-14:
How can I publish a data from the MQTT broker to the MobdusPal simulator?

cloudxxx8, 2020-12-15:
Please use app service to push data to MQTT broker

CesarCastillo-90, 2020-12-15:
Hi mr. @cloudxxx8, ok, but I don't know how do it. Could you guide me in this?, I'm using docker compose Hanoi.

[docker-compose-hanoi-arm64.zip](https://github.com/edgexfoundry/device-modbus-go/files/5696442/docker-compose-hanoi-arm64.zip)

weichou1229, 2020-12-16:
This might works for you https://docs.edgexfoundry.org/1.3/microservices/application/AppServiceConfigurable/

CesarCastillo-90, 2020-12-16:
ok mr. @weichou1229, I'm going to check, thank you so much!