edgexfoundry/device-modbus-go #220: DeviceResource  PropertyValue data-transformation

VijayKumarGopu, 2021-03-06:
Hi,

Sensor SHT20 sends the temperature value as multiple of 10 with modbus protocol. 

For example 30.3 degree celsius is sent as 303.
----------------------------------------------------------------------------
modbus: sending 01 04 00 01 00 01 60 0a
modbus: received 01 04 02 01 2f f9 7c
----------------------------------------------------------------------------

With the following device resource I can only get the temperature value as 30 instead of 30.3
----------------------------------------------------------------------------
deviceResources:
      -
        name: "TemperatureDegC"
        description: "Room Temperature in Degree Celsius."
        attributes:
            {  primaryTable: "INPUT_REGISTERS", startingAddress: "2" }
        properties:
            value:
                { type: "Int16", readWrite: "R", size: "1", scale: "0.1", minimum: "-40.0", maximum: "60.0", defaultValue: "0"}
            units:
                { type: "String", readWrite: "R", defaultValue: "degree celsius"}
----------------------------------------------------------------------------

Output:
----------------------------------------------------------------------------
{
  "device": "TemperatureHumiditySensor",
  "origin": 1615019852716369000,
  "readings": [
    {
      "origin": 1615019852715212000,
      "device": "TemperatureHumiditySensor",
      "name": "TemperatureDegC",
      "value": "31",
      "valueType": "Int16"
    }
  ],
  "EncodedEvent": null
}
----------------------------------------------------------------------------

I could not see a way to convert from INT value to FLAOT and apply a scale of 0.1.

Please advise.

Thanks
Vijay Kumar

cloudxxx8, 2021-03-07:
@VijayKumarGopu please see this document
https://docs.edgexfoundry.org/1.3/examples/Ch-ExamplesModbusdatatypeconversion/

We have resolved this case by the new attribute "rawType"

VijayKumarGopu, 2021-03-07:
thanks for the info @cloudxxx8. With rawType I can retain the sensor values.

Thanks
Vijay Kumar

cloudxxx8, 2021-03-07:
No problem.  I am glad you can.

closed 2021-03-07 by cloudxxx8