edgexfoundry/device-modbus-go #232: Modbus example error - connection refused

stlaperriere, 2021-05-07:
Hi,

I am trying (and so far failing) to follow the Modbus example: https://github.com/edgexfoundry/edgex-docs/blob/fuji/examples/Ch-ExamplesAddingModbusDevice.rst#id1

I am running the example on a Windows machine. ModbusPal runs natively on Windows and Docker containers run in WSL2 (Ubuntu 20.04). 

To simplify things, I am using release Fuji, since this version is explicitly mentioned in the above documentation. 

Here are the steps I followed:
* _Configure_ and _start_ ModbusPal simulation as per the documentation
* Copy the Fuji docker-compose.yml onto my WSL
* Uncomment service `device-modbus` in the compose file (then I left it as-is)
* `docker-compose up`
* Add the device profile using `curl localhost:48081/api/v1/deviceprofile/uploadfile -F "file=@DENT.Mod.PS6037.profile.yaml"`
* Add the device using `curl localhost:48081/api/v1/device -H "Content-Type:application/json" -X POST \
  -d '{
   "name" :"Modbus-TCP-Device",
   "description":"This device is a product for monitoring and controlling digital inputs and outputs over a LAN.",
   "adminState":"UNLOCKED",
   "operatingState":"ENABLED",
   "protocols":{
      "modbus-tcp":{
         "Address" : "172.17.0.1",
         "Port" : "1502",
         "UnitID" : "1"
      }
   },
   "labels":[
      "Air conditioner",
      "modbus TCP"
   ],
   "service":{
      "name":"edgex-device-modbus",
      "adminState": "unlocked",
      "operatingState": "enabled"
   },
   "profile":{
      "name":"Network Power Meter"
   }
}'`

All of the above commands seem to work. Next step is to try to read some values from the simulated device using `curl --location --request GET 'http://localhost:48082/api/v1/device/name/Modbus-TCP-Device/command/Configuration'`

I get the following error : `Handler - execReadCmd: error for Device: Modbus-TCP-Device cmd: Configuration, dial tcp 172.17.0.1:1502: connect: connection refused /api/v1/device/11eb8edf-6f45-44e1-b673-8729a5645d3e/Configuration`

Am I missing something?

weichou1229, 2021-05-08:
Did you click the `Run` button for running the ModbusPal?

>  The only thing I can think of is maybe WSL does not forward the connections or the ports adequately?

Yes. Since you run the ModbusPal on the Windows and the edgex run in the WSL2, so the 172.17.0.1(docker bridge network) might not works. You might need to use the windows host IP like 192.x.x.x

stlaperriere, 2021-05-10:
Thank you for the quick reply. Yes, the simulation was running in ModbusPal.

It looks like the problem was indeed linked to WSL2. I tried on a native Linux machine using the exact same steps (and leaving the IP address as `172.17.0.1`) and it worked.