edgexfoundry/device-modbus-go #287: Make docker Error

bighb69738, 2021-11-02:
OS: mac
GO version: go version go1.16.3 darwin/amd64

I tried to use the command "go mod tidy", but it still fails.

[image]

weichou1229, 2021-11-02:
I downloaded the repo and run `make docker` but didn't find any issue, maybe you can try to clean the go mod cache or clone the repo again.
```
[+] Building 107.5s (21/21) FINISHED                                                                                                                                  
 => [internal] load build definition from Dockerfile                                                                                                             0.1s
 => => transferring dockerfile: 1.65kB                                                                                                                           0.0s
 => [internal] load .dockerignore                                                                                                                                0.0s
 => => transferring context: 265B                                                                                                                                0.0s
 => [internal] load metadata for docker.io/library/alpine:3.14                                                                                                   3.4s
 => [internal] load metadata for docker.io/library/golang:1.16-alpine3.14                                                                                        2.4s
 => [auth] library/golang:pull token for registry-1.docker.io                                                                                                    0.0s
 => [auth] library/alpine:pull token for registry-1.docker.io                                                                                                    0.0s
 => CACHED [stage-1 1/6] FROM docker.io/library/alpine:3.14@sha256:e1c082e3d3c45cccac829840a25941e679c25d438cc8412c2fa221cf1a824e6a                              0.0s
 => [internal] load build context                                                                                                                                0.1s
 => => transferring context: 153.39kB                                                                                                                            0.1s
 => CACHED [builder 1/7] FROM docker.io/library/golang:1.16-alpine3.14@sha256:05df7ff684a2cb06aa207be14a78918cbc3285ed3b965974979e575d59de1c11                   0.0s
 => [builder 2/7] RUN sed -e 's/dl-cdn[.]alpinelinux.org/nl.alpinelinux.org/g' -i~ /etc/apk/repositories                                                         0.7s
 => [stage-1 2/6] RUN sed -e 's/dl-cdn[.]alpinelinux.org/nl.alpinelinux.org/g' -i~ /etc/apk/repositories                                                         0.7s
 => [stage-1 3/6] RUN apk add --update --no-cache zeromq dumb-init                                                                                               7.1s
 => [builder 3/7] RUN apk add --update --no-cache make git openssh gcc libc-dev zeromq-dev libsodium-dev                                                        73.9s
 => [builder 4/7] WORKDIR /device-modbus-go                                                                                                                      0.0s 
 => [builder 5/7] COPY . .                                                                                                                                       0.0s 
 => [builder 6/7] RUN [ ! -d "vendor" ] && go mod download all || echo "skipping..."                                                                            20.6s 
 => [builder 7/7] RUN make build                                                                                                                                 8.5s 
 => [stage-1 4/6] COPY --from=builder /device-modbus-go/cmd /                                                                                                    0.1s 
 => [stage-1 5/6] COPY --from=builder /device-modbus-go/LICENSE /                                                                                                0.0s 
 => [stage-1 6/6] COPY --from=builder /device-modbus-go/Attribution.txt /                                                                                        0.0s 
 => exporting to image                                                                                                                                           0.1s 
 => => exporting layers                                                                                                                                          0.1s
 => => writing image sha256:b557cf2cf70358b7229e9284f44f8900a5891d27fcd91c3937232d77a8d9548a                                                                     0.0s
 => => naming to docker.io/edgexfoundry/device-modbus:0.0.0-dev
```

mac os version: 11.6
go version: go1.16.8 darwin/amd64

bighb69738, 2021-11-02:
@weichou1229 
Thank you for your reply!  
After I tried to clean the go mod cache and clone the repo again, it can work now.

closed 2021-11-02 by bighb69738