edgexfoundry/device-modbus-go #288: Device command Error

bighb69738, 2021-11-02:
OS: mac 11.6
Go: go version go1.16.3 darwin/amd64
I built the docker image and started up EdgeX by docker-compose.
When i use device command, but it fails.
[image]

**docker-compose** 
[image]

**cmd/configuration.toml**
[Writable]
LogLevel = "INFO"
  # InsecureSecrets are required for when Redis is used for message bus
  [Writable.InsecureSecrets]
    [Writable.InsecureSecrets.Sample]
    path = "redisdb"
      [Writable.InsecureSecrets.Sample.Secrets]
      username = ""
      password = ""

[Service]
HealthCheckInterval = "10s"
Host = "localhost"
Port = 59901
ServerBindAddr = ""  # blank value defaults to Service.Host value
StartupMsg = "device modbus started"
# MaxRequestSize limit the request body size in byte of put command
MaxRequestSize = 0 # value 0 unlimit the request size.
RequestTimeout = "5s"
  [Service.CORSConfiguration]
  EnableCORS = false
  CORSAllowCredentials = false
  CORSAllowedOrigin = "https://localhost"
  CORSAllowedMethods = "GET, POST, PUT, PATCH, DELETE"
  CORSAllowedHeaders = "Authorization, Accept, Accept-Language, Content-Language, Content-Type, X-Correlation-ID"
  CORSExposeHeaders = "Cache-Control, Content-Language, Content-Length, Content-Type, Expires, Last-Modified, Pragma, X-Correlation-ID"
  CORSMaxAge = 3600

[Registry]
Host = "localhost"
Port = 8500
Type = "consul"

[Clients]
  [Clients.core-data]
  Protocol = "http"
  Host = "localhost"
  Port = 59880

  [Clients.core-metadata]
  Protocol = "http"
  Host = "localhost"
  Port = 59881

[MessageQueue]
Protocol = "redis"
Host = "localhost"
Port = 6379
Type = "redis"
AuthMode = "usernamepassword"  # required for redis messagebus (secure or insecure).
SecretName = "redisdb"
PublishTopicPrefix = "edgex/events/device" # /<device-profile-name>/<device-name>/<source-name> will be added to this Publish Topic prefix
  [MessageQueue.Optional]
  # Default MQTT Specific options that need to be here to enable environment variable overrides of them
  # Client Identifiers
  ClientId = "device-modbus"
  # Connection information
  Qos = "0" # Quality of Sevice values are 0 (At most once), 1 (At least once) or 2 (Exactly once)
  KeepAlive = "10" # Seconds (must be 2 or greater)
  Retained = "false"
  AutoReconnect = "true"
  ConnectTimeout = "5" # Seconds
  SkipCertVerify = "false" # Only used if Cert/Key file or Cert/Key PEMblock are specified  [MessageQueue.Optional]

# Only used when EDGEX_SECURITY_SECRET_STORE=true which is now required for secure Consul
[SecretStore]
Type = "vault"
Host = "localhost"
Port = 8200
Path = "device-modbus/"
Protocol = "http"
RootCaCertPath = ""
ServerName = ""
SecretsFile = ""
DisableScrubSecretsFile = false
TokenFile = "/tmp/edgex/secrets/device-modbus/secrets-token.json"
  [SecretStore.Authentication]
  AuthType = "X-Vault-Token"

[Device]
DataTransform = true
MaxCmdOps = 128
MaxCmdValueLen = 256
ProfilesDir = "./res/profiles"
DevicesDir = "./res/devices"
UpdateLastConnected = false
Labels = []
EnableAsyncReadings = true
AsyncBufferSize = 16
UseMessageBus = true

cloudxxx8, 2021-11-02:
the Modbus simulator is not accessible by the Modbus Device Service.  Please use the docker0 IP instead of 0.0.0.0

bighb69738, 2021-11-03:
@cloudxxx8 
Thanks

closed 2021-11-03 by bighb69738