edgexfoundry/device-modbus-go #290: Set up the Modbus RTU Device but use the TCP protocol ?

bighb69738, 2021-11-03:
Set up the Modbus RTU Device and create the device entity to the EdgeX. 
https://docs.edgexfoundry.org/2.0/examples/Ch-ExamplesAddingModbusDevice/

There is a little weird about the protocol setting.

---
"labels":[ 
                  "IO Module",
                  "Modbus RTU"
               ],
               "serviceName": "device-modbus",
               "profileName": "Ethernet-Temperature-Sensor",
               "protocols":{
                  "**modbus-tcp**":{
                     "Address" : "/dev/ttyUSB0",
                     "BaudRate" : "19200",
                     "DataBits" : "8",
                     "StopBits" : "1",
                     "Parity" : "N",
                     "UnitID" : "1",
                     "Timeout" : "5",
                     "IdleTimeout" : "5"
                  }
  },

---
My question is about why we set up the Modbus RTU Device but use the TCP protocol ??

weichou1229, 2021-11-03:
It's a typo, you can find the correct example config at https://github.com/edgexfoundry/device-modbus-go/blob/main/cmd/res/devices/modbus.test.devices.toml#L25

bighb69738, 2021-11-04:
@weichou1229 
Thank you!

closed 2021-11-04 by bighb69738