edgexfoundry/device-modbus-go #293: Fail to create read command connection info.

bighb69738, 2021-11-11:
docker logs 
[image]

docker-compose 
[image]

device setting
[image]

I tried to use the minimalmodbus to get the readings from the sensor by RTU and that can work.

weichou1229, 2021-11-11:
Yes, you can to define the protocol properties according to the sample https://github.com/edgexfoundry/device-modbus-go/blob/main/cmd/res/devices/modbus.test.devices.toml#L25

bighb69738, 2021-11-12:
@weichou1229  
Thank you. I hope you have a good day.

closed 2021-11-12 by bighb69738