edgexfoundry/device-modbus-go #300: Use of the internal package is not allowed

1486756632, 2021-11-29:
##
[image]
# 
i use goland to compile code. Why there is always a error "Use of the internal package is not allowed". I pull the code from github. I do not change any code. Is there anyone could help me.Thanks a lot

cloudxxx8, 2021-11-29:
that's the design of golang.  you can't use the internal package of other repository.
https://go.dev/doc/go1.4#internalpackages

1486756632, 2021-11-30:
> that's the design of golang. you can't use the internal package of other repository. https://go.dev/doc/go1.4#internalpackages

There is a question.How can run the code ?How can i debug the code?

cloudxxx8, 2021-11-30:
no, you can't use the internal package from another repository.

1486756632, 2021-11-30:
> repository

Actually, i wanna rebuild the code.Would u mind tell me ,if i rebuild the code .How could i run main.go by GoLand.

cloudxxx8, 2021-11-30:
from the screenshot, you are developing your code from another repository.
you have to clone this repository and put your main.go into it, and it will work.

lenny-intel, 2022-08-11:
@1486756632 , can this issue be closed?