edgexfoundry/device-modbus-go #304: Modbus-TCP-Device not found

ngocnguyenbk12, 2022-01-15:
Hi, I just start with edgex foundry
i done with random number device
now i move to Modbus.
i do follow the instruction, i see my Modbus Device when I GET: localhost:48082/api/v1/device:
 {
        "id": "169e8e92-25f1-429d-b587-6b30d8b3f4f4",
        "name": "Modbus-TCP-Device",
        "adminState": "UNLOCKED",
        "operatingState": "ENABLED",
        "labels": [
            "Air conditioner",
            "modbus TCP"
        ],
        "commands": [
            {
                "created": 1642232444536,
                "modified": 1642232444536,
                "id": "f76a8303-d035-46d5-9672-99f83dffacba",
                "name": "Configuration",
                "get": {
                    "path": "/api/v1/device/{deviceId}/Configuration",
                    "responses": [
                        {
                            "code": "200",
                            "description": "Get the Configuration",
                            "expectedValues": [
                                "DemandWindowSize",
                                "LineFrequency"
                            ]
                        },
                        {
                            "code": "500",
                            "description": "internal server error"
                        }
                    ],
                    "url": "http://edgex-core-command:48082/api/v1/device/169e8e92-25f1-429d-b587-6b30d8b3f4f4/command/f76a8303-d035-46d5-9672-99f83dffacba"
                },

but when i try to get value from GET: http://localhost:48082/api/v1/device/169e8e92-25f1-429d-b587-6b30d8b3f4f4/command/f76a8303-d035-46d5-9672-99f83dffacba

And it show me this:

Configuration for Device: Modbus-TCP-Device not found; GET /api/v1/device/169e8e92-25f1-429d-b587-6b30d8b3f4f4/Configuration

Can U guys help me,
Thank you so much.

cloudxxx8, 2022-01-17:
If you are a new starter, could you please try to use the latest version?  It is v2.1 now, and it is the long term support version.

ngocnguyenbk12, 2022-01-20:
aw, lemme try at newer version

lenny-intel, 2022-08-11:
@ngocnguyenbk12 , can we close this issue?

iain-anderson, 2022-09-07:
closing as inactive

closed 2022-09-07 by iain-anderson