edgexfoundry/device-modbus-go #353: Device Modbus get null deviceName when updating serviceName from device-mqtt to device-modbus

Gaurang2811, 2022-06-16:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

edgex-device-modbus
 The issue is located in: edgex-device-modbus

### Is this a regression?

 No

### Description and Minimal Reproduction [**REQUIRED**]
If we create a new device with the serviceName as `device-mqtt` and then update the serviceName from `device-mqtt` to `device-modbus`, then the deviceName in **modbus** has a null value.
Modbus get the correct value of deviceName and works fine after restart of **edgex-device-modbus**.

I have created a device profile( name: PROFILE) with a single resource( name: RESOURCE), and a device( name: DEVICE).
Initially, I created the device with the service `device-mqtt` and then updated the service from `device-mqtt` to `device-modbus`.

Redis pattern before **edgex-device-modbus** restart: `edgex.events.device.PROFILE..RESOURCE`
Device-Modbus debug logs before **edgex-device-modbus** restart:
`level=DEBUG ts=2022-06-16T11:00:51.62387229Z app=device-modbus source=utils.go:73 msg="Event(profileName: PROFILE, deviceName: , sourceName: RESOURCE, id: c5919d73-2287-4d30-a154-25309695f66e) published to MessageBus"
`

Redis pattern after **edgex-device-modbus** restart: `edgex.events.device.PROFILE.DEVICE.RESOURCE`
Device-Modbus debug logs after **edgex-device-modbus** restart:
`level=DEBUG ts=2022-06-16T11:01:27.166122328Z app=device-modbus source=utils.go:73 msg="Event(profileName: PROFILE, deviceName: DEVICE, sourceName: RESOURCE, id: 70d3cb49-5000-483f-a0be-228480969a23) published to MessageBus"
`

Steps to reproduce.
Create any device with device service `device-mqtt`.
then update the device and change the device service from `device-mqtt` to `device-modbus`

## 🔥 Exception or Error
<pre><code>

</code></pre>

## 🌍  Your Environment
**Deployment Environment:** Linux x86-64

**EdgeX Version [**REQUIRED**]:** Jakarta( 2.1)

hahattan, 2022-07-29:
@Gaurang2811 Could you verify the latest SDK change solves this issue ?
https://github.com/edgexfoundry/device-sdk-go/commit/ff1b5b02c8f314790bf5be564bfc978a145e3053

Gaurang2811, 2022-08-16:
Can we merge this fix in the Jakarta release?

cloudxxx8, 2022-08-16:
@Gaurang2811 we could merge this to the Jakarta branch of device-sdk-go, but you still need to build your own images.
I suggest you just building your own images with the latest device-sdk-go

iain-anderson, 2022-10-24:
Assuming this is fixed in the current codebase - closing.

closed 2022-10-24 by iain-anderson