edgexfoundry/device-modbus-go #356: (Modbus TCP)device-modbus-go using multi connections, but real devices support one connection only.

zhurichard, 2022-06-21:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 
[device-modbus-go](Modbus TCP)

### Is this a regression?

 No, all versions have the same issue.

### Description and Minimal Reproduction [**REQUIRED**]
Device-modbus-go connects modbus during HandleReadCommands、HandleWriteCommands in driver.go, This will client muti DeviceClient instances.  

So that the "connect()" function in tcpclient.go(github.com/goburrow/modbus) be invoked multi times before the previours connection closed. 

It is ok for some of the devices which have ability of muti tcp connections, but lot of others only support one tcp connection. that will get error showing in Exception section.

Hope this can be solved. At least to give one setting to config to use one tcp connection or muti tcp connections. May be it has the functionality already, but I don't know? 

## 🔥 Exception or Error
<pre><code>

code: 500 , message: request failed, status code: 500, err: {"apiVersion":"v2","message":"error reading DeviceResourece TemperatureTop for TCP-Temperature-Humidity -\u003e dial tcp 192.168.1.17:502: connect: connection refused","statusCode":500}
</code></pre>

## 🌍  Your Environment
**Deployment Environment:**
raspiberry pi 4

**EdgeX Version [**REQUIRED**]:**
all versions

**Anything else relevant?**

weichou1229, 2022-06-21:
The device-modbus-go already implemented that by using the lock, please check whether the device is connectable.
https://github.com/edgexfoundry/device-modbus-go/blob/4c89d6a6c91720700353e870f5243ca140b0cc6d/internal/driver/driver.go#L97

jiekechoo, 2023-02-12:
You could add a `secondary ip address` in modbus slave. In my test, there are 200 ip address in slave.