edgexfoundry/device-modbus-go #367: How to create the device-modbus service with docker swarm?

bighb69738, 2022-08-09:
Hi all:
I tried to create the device-modbus service with docker swarm, but fail.
The errror message is "operation not permitted".
I tried to use "cap_add -all" and "propagation: shared", but it still fail to open "/dev/ttymexc1".

### docker-compose.yml
```
...
device-modbus:
    depends_on:
      - consul
      - data
      - metadata
    environment:
      CLIENTS_CORE_COMMAND_HOST: edgex-core-command
      CLIENTS_CORE_DATA_HOST: edgex-core-data
      CLIENTS_CORE_METADATA_HOST: edgex-core-metadata
      CLIENTS_SUPPORT_NOTIFICATIONS_HOST: edgex-support-notifications
      CLIENTS_SUPPORT_SCHEDULER_HOST: edgex-support-scheduler
      DATABASES_PRIMARY_HOST: edgex-redis
      EDGEX_SECURITY_SECRET_STORE: "false"
      MESSAGEQUEUE_HOST: edgex-redis
      REGISTRY_HOST: edgex-core-consul
      SERVICE_HOST: edgex-device-modbus
    hostname: edgex-device-modbus
    image: edgexfoundry/device-modbus-arm64:2.0.0
    networks:
      edgex-network: {}
    ports:
      - target: 59901
        published: 59901
        protocol: tcp
        mode: host
    volumes:
      - type: bind
        source: /dev/ttymxc1
        target: /dev/ttymxc1
        bind:
          propagation: shared
    user: root:root
    cap_add:
      - All
    deploy:
      placement:
        constraints:
        - "node.id==yb6n8699xhia34ujka9ss7m34"
...
```

### docker inspect the device-modbus
```
[
    {
        "Id": "ef4b8085f0765572391868ebcd615e4f0835e542fbfcaf58d026bdf471d97f95",
        "Created": "2022-08-09T01:10:17.617447532Z",
        "Path": "/device-modbus",
        "Args": [
            "--cp=consul://edgex-core-consul:8500",
            "--registry",
            "--confdir=/res"
        ],
        "State": {
            "Status": "running",
            "Running": true,
            "Paused": false,
            "Restarting": false,
            "OOMKilled": false,
            "Dead": false,
            "Pid": 7805,
            "ExitCode": 0,
[... 227 lines cut ...]
                "example_edgex-network": {
                    "IPAMConfig": {
                        "IPv4Address": "10.0.2.33"
                    },
                    "Links": null,
                    "Aliases": [
                        "ef4b8085f076",
                        "edgex-device-modbus"
                    ],
                    "NetworkID": "ag8t5o6ud8bu18lbwdq00j5jt",
                    "EndpointID": "883c5399ec793818bf064a94ad33dcec8111596b743175208ce46d4f6d92d217",
                    "Gateway": "",
                    "IPAddress": "10.0.2.33",
                    "IPPrefixLen": 24,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "02:42:0a:00:02:21",
                    "DriverOpts": null
                }
            }
        }
    }
]
```

weichou1229, 2022-08-09:
The 

> I tried to use "cap_add -all" and "propagation: shared", but it still fail to open "/dev/ttymexc1".

You can try this doc https://docs.edgexfoundry.org/2.2/examples/Ch-ExamplesAddingModbusDevice/#set-up-the-modbus-rtu-device

bighb69738, 2022-08-09:
Hi @weichou1229 :

Thank you for your reply.
But "device_cgroup_rules" can not be used by docker stack deploy.

weichou1229, 2022-08-09:
> But "device_cgroup_rules" can not be used by docker stack deploy.

Sorry, I am not familiar with the docker swarm deployment.

lindseysimple, 2023-07-10:
Closing this issue as the Docker Swarm deployment strategy is out of the scope of EdgeX supported deployment methods.