edgexfoundry/device-modbus-go #369: MODBUS - device-modbus-go

boaretti, 2022-08-24:
Hi,
Please, when reading a single HOLDING_REGISTER is it possible to decouple a single read in multiple
values. My case a single register the information represented as BITs so If possible I would like to derive several information
from this single read. Is it possible ? Could someone point me to the correct place ?

cloudxxx8, 2022-08-25:
Yes, it's possible by modifying the driver code.  You can feel free to open a PR for this mechanism.

jiekechoo, 2023-02-12:
@cloudxxx8 
I have a modbus tcp slave, it has  >5 register data, is it possible to `couple` to one event in `AutoEvents`?

> Yes, it's possible by modifying the driver code. You can feel free to open a PR for this mechanism.

cloudxxx8, 2023-02-13:
Yes, you define one Device Command in the Device Profile to aggregate multiple Device Resources (5 in your case).
https://docs.edgexfoundry.org/2.3/walk-through/Ch-WalkthroughDeviceProfile/#understanding-device-commands

jiekechoo, 2023-02-13:
@cloudxxx8 
[image]
got it, thanks.