edgexfoundry/device-modbus-go #531: when send write-commond with device-modbus-rtu, errors appears

Liguangwei0321, 2023-12-08:
using EdgeX verison 3.1, and Connecting to a Modbus Device via RS232   read succussfully,but  with write commond error will appear :
code: 500 , message: request failed, status code: 500, err: {"apiVersion":"v3","message":"error writing DeviceCommand CS0 state for Modbus-RTU-IO-Module -\u003e handle write command request failed, err: modbus: exception '1' (illegal function), function '143'","statusCode":500}

detail: define
- description: On/Off , 0-OFF 1-ON
      name: CS100
      isHidden: false
      properties:
        valueType: Bool
        readWrite: RW
        units: ""
        minimum: null
        maximum: null
        defaultValue: ""
        mask: null
        shift: null
        scale: null
        offset: null
        base: null
        assertion: ""
        mediaType: ""
        optional: {}
      attributes:
        primaryTable: COILS
        startingAddress: 100

jiekechoo, 2023-12-08:
@Liguangwei0321 Here is an issue you can refer: https://github.com/edgexfoundry/device-modbus-go/issues/167

Liguangwei0321, 2023-12-08:
> @Liguangwei0321这里有一个问题你可以参考：#167

error code canot  match，i see：write the HOLDING_REGISTERS successfully，but write  COILS  failed

Liguangwei0321, 2023-12-08:
using docker  logs，get the infomation when write Coils commond:
edgex-device-modbus                    | level=ERROR ts=2023-12-08T08:13:03.737765793Z app=device-modbus source=driver.go:194 msg="handle write command request failed, err: modbus: exception '1' (illegal function), function '143'"
edgex-device-modbus                    | level=ERROR ts=2023-12-08T08:13:03.742922371Z app=device-modbus source=restrouter.go:159 X-Correlation-ID=668a39c8-de5d-4249-a315-de0d00e25272 msg="error writing DeviceResource CS100 for Modbus-RTU-IO-Module -> handle write command request failed, err: modbus: exception '1' (illegal function), function '143'"
edgex-core-command                     | level=ERROR ts=2023-12-08T08:13:03.744185904Z app=core-command source=http.go:47 X-Correlation-ID=df741648-23c9-424e-8e12-0bf53d664230 msg="request failed, status code: 500, err: {\"apiVersion\":\"v3\",\"message\":\"error writing DeviceResource CS100 for Modbus-RTU-IO-Module -\\u003e handle write command request failed, err: modbus: exception '1' (illegal function), function '143'\",\"statusCode\":500}"
edgex-device-modbus                    | 2023/12/08 16:13:06 modbus: closing connection due to idle timeout: 3.000694022s
edgex-device-modbus                    | level=INFO ts=2023-12-08T08:13:09.664131798Z app=device-modbus source=modbusclient.go:38 msg="Modbus client create TCP connection."
edgex-device-modbus                    | level=INFO ts=2023-12-08T08:13:09.664241815Z app=device-modbus source=driver.go:116 msg="Read command OpenConnection failed. err:dial tcp 192.168.1.120:502: connect: connection refused \n"
edgex-device-modbus                    | level=ERROR ts=2023-12-08T08:13:09.664354443Z app=device-modbus source=executor.go:55 msg="AutoEvent - error occurs when reading resource Temperature: error reading Regex DeviceResource(s) Temperature for Modbus-TCP-Temperature-Sensor -> dial tcp 192.168.1.120:502: connect: connection refused"
edgex-core-consul                      | 2023-12-08T08:13:15.709Z [INFO]  agent: Synced check: check=support-notifications

Liguangwei0321, 2023-12-08:
> **edgex-device-modbus | 2023/12/08 16:19:15 modbus: received 01 8f 01 85 f0**

> @Liguangwei0321 Here is an issue you can refer: #167

i get more logs ,can you help me

jiekechoo, 2023-12-08:
> > **edgex-device-modbus | 2023/12/08 16:19:15 modbus: received 01 8f 01 85 f0**
> 
> > @Liguangwei0321 Here is an issue you can refer: #167
> 
> i get more logs ,can you help me

Please change the loglevel to `DEBUG`, and try again.

jiekechoo, 2023-12-08:
> > DEBUG
> 
> 
> 
> how to run debug mode

There are two options: 
- Modify `configuration.yaml` , Writable-LogLevel: INFO->DEBUG
- Add an environment: WRITABLE_LOGLEVEL:DEBUG

Execute command `make clean` before starting docker.

Liguangwei0321, 2023-12-11:
> > > 调试
> > 
> > 
> > 如何运行调试模式
> 
> 有两种选择：
> 
> * 修改`configuration.yaml`，可写日志级别：INFO->DEBUG
> * 添加环境：WRITABLE_LOGLEVEL:DEBUG
> 
> `make clean`在启动 docker 之前执行命令。
I noticed that the code in version 3.1 defaults to using WriteMultipleCoils without checking if multiple registers are being written. I am writing to a single register. Should I consider using the WriteSingleCoil function?

lindseysimple, 2023-12-12:
Assigning to @jiekechoo for further investigation.

jiekechoo, 2023-12-13:
> Assigning to @jiekechoo for further investigation.

I'm working on it.

jiekechoo, 2023-12-13:
@Liguangwei0321 Could tell me what's your modbus slave device type? and please provide the device user guide.

I just use `Modbus Slave` as a modbus-tcp simlulator, the device service works.

Liguangwei0321, 2023-12-14:
> @Liguangwei0321 Could tell me what's your modbus slave device type? and please provide the device user guide.
> 
> I just use `Modbus Slave` as a modbus-tcp simlulator, the device service works.

i use the  modbus rtu with rs232toUSB，and  connect to a modbus rtu device，

cloudxxx8, 2023-12-16:
According to the log, the modbus device service sends the correct request, but got the illegal function result from the device
https://www.simplymodbus.ca/exceptions.htm

Is there a chance that device doesn't support the Multiple Coils write?
https://www.simplymodbus.ca/FC15.htm

Are you able to send the Multiple Coils write from another software to verify it?

jiekechoo, 2023-12-16:
@cloudxxx8 It depends on the modbus slave device's protocol. I think, @Liguangwei0321 you should provide the device user guide.

Liguangwei0321, 2023-12-19:
> @cloudxxx8这取决于 modbus 从设备的协议。我认为，@Liguangwei0321您应该提供设备用户指南。

I modified the operation for writing COILS in the SDK, adding a check for individual registers, but there is still an issue with the writing. The modified code is as follows:
**case COILS:
		if modbusCommandInfo.Length == 1 {
			result, err = c.client.WriteSingleRegister(uint16(modbusCommandInfo.StartingAddress), binary.BigEndian.Uint16(value))
		} else {
			result, err = c.client.WriteMultipleCoils(uint16(modbusCommandInfo.StartingAddress), modbusCommandInfo.Length, value)
		}**

error info：
2023/12/19 09:34:30 http: panic serving 172.19.0.12:44036: runtime error: index out of range [1] with length 1
goroutine 179 [running]:
net/http.(*conn).serve.func1()
        net/http/server.go:1868 +0xb0
panic({0x963340?, 0x40003e0420?})
        runtime/panic.go:920 +0x26c
net/http.(*timeoutHandler).ServeHTTP(0x40000310c0, {0xb279e8, 0x4000590a60}, 0x4000253d00)
        net/http/server.go:3402 +0x6bc
github.com/edgexfoundry/go-mod-bootstrap/v3/bootstrap/handlers.(*HttpServer).BootstrapHandler.TimeoutWithConfig.TimeoutConfig.ToMiddleware.func9.1({0xb38f18, 0x4000498000})
        github.com/labstack/echo/v4@v4.11.2/middleware/timeout.go:125 +0x1ac
github.com/edgexfoundry/go-mod-bootstrap/v3/bootstrap/handlers.(*HttpServer).BootstrapHandler.UrlDecodeMiddleware.func5.1({0xb38f18, 0x4000498000})
        github.com/edgexfoundry/go-mod-bootstrap/v3@v3.1.0/bootstrap/handlers/common_middleware.go:84 +0x260
github.com/edgexfoundry/go-mod-bootstrap/v3/bootstrap/handlers.(*HttpServer).BootstrapHandler.LoggingMiddleware.func4.1({0xb38f18, 0x4000498000})
        github.com/edgexfoundry/go-mod-bootstrap/v3@v3.1.0/bootstrap/handlers/common_middleware.go:60 +0x3b0
github.com/edgexfoundry/go-mod-bootstrap/v3/bootstrap/handlers.ManageHeader.func1({0xb38f18, 0x4000498000})
        github.com/edgexfoundry/go-mod-bootstrap/v3@v3.1.0/bootstrap/handlers/common_middleware.go:40 +0x228
github.com/labstack/echo/v4.(*Echo).ServeHTTP(0x40001b0b40, {0xb27808?, 0x4000393340}, 0x4000253c00)
        github.com/labstack/echo/v4@v4.11.2/echo.go:669 +0x374
net/http.serverHandler.ServeHTTP({0xb257f0?}, {0xb27808?, 0x4000393340?}, 0x6?)
        net/http/server.go:2938 +0xbc
net/http.(*conn).serve(0x400018c900, {0xb28cc8, 0x40004dc3c0})
        net/http/server.go:2009 +0x518
created by net/http.(*Server).Serve in goroutine 12
        net/http/server.go:3086 +0x4cc
2023/12/19 09:34:31 modbus: closing connection due to idle timeout: 5.000498873s

jiekechoo, 2023-12-19:
@Liguangwei0321 , you have to provide your device profile, device configruation, and the modbus slave user mannual.

In some modbus rtu device, you only write coils with `FF00` as a `ON` and `0000` as `OFF`. So, it depends on your modbus slave device.

Liguangwei0321, 2023-12-19:
> @Liguangwei0321 , you have to provide your device profile, device configruation, and the modbus slave user mannual.
> 
> In some modbus rtu device, you only write coils with `FF00` as a `ON` and `0000` as `OFF`. So, it depends on your modbus slave device.

但是这次错误是虚拟设备导致的，http: panic serving 172.19.0.12:44036: runtime error: index out of range [1] with length 1
goroutine 179，并非modbus从设备

Liguangwei0321, 2023-12-20:
> @cloudxxx8 It depends on the modbus slave device's protocol. I think, @Liguangwei0321 you should provide the device user guide.
[modbus设备协议.zip](https://github.com/edgexfoundry/device-modbus-go/files/13722141/modbus.zip)
设备协议见附件

jiekechoo, 2023-12-20:
> Are you able to send the Multiple Coils write from another software to verify it?

Did you try this?

Liguangwei0321, 2023-12-20:
我通过修改sdk中的单独写COILs解决了该问题，result, err = c.client.WriteSingleRegister(uint16(modbusCommandInfo.StartingAddress), binary.BigEndian.Uint16(value))
但是数据在进行BigEndian.Uint16时，需要将单字节变更为双字节没否则会报错

closed 2023-12-20 by Liguangwei0321