edgexfoundry/device-modbus-go #532: How to write yaml for device-modbus

Liguangwei0321, 2023-12-12:
在编译SDK verison 3.1 napa时，yaml报错，提示不完整；
1.请问如何书写yaml，填写哪些必要元素
2.为什么2.0版本有完整的res/toml文件，3.1版本的yaml有很多空缺，没有完整示例

jiekechoo, 2023-12-14:
`level=ERROR ts=2023-12-13T07:07:51.173281435Z app=device-modbus source=messaging.go:45 msg="MessageBus configuration is incomplete, missing common config? Use -cp or -cc flags for common config."`

Here is a guide: https://docs.edgexfoundry.org/3.1/microservices/configuration/CommonCommandLineOptions/ you could refer.

Try this command: 
```
./device-modbus -cp -d -o
```

Liguangwei0321, 2023-12-18:
> `level=ERROR ts=2023-12-13T07:07:51.173281435Z app=device-modbus source=messaging.go:45 msg="MessageBus configuration is incomplete, missing common config? Use -cp or -cc flags for common config."`
> 
> 这里有一个指南：https://docs.edgexfoundry.org/3.1/microservices/configuration/CommonCommandLineOptions/您可以参考。
> 
> 尝试这个命令：
> 
> ```
> ./device-modbus -cp -d -o
> ```

使用你提示的命令，依然有错误：
level=ERROR ts=2023-12-18T01:47:29.451021229Z app=device-modbus source=messaging.go:45 msg="MessageBus configuration is incomplete, missing common config? Use -cp o
[image]

closed 2023-12-18 by Liguangwei0321