edgexfoundry/device-modbus-go #535: when Executing read commod, permission denied

Liguangwei0321, 2023-12-19:
When I use the modbus-SDK to perform 'make build', compile, and then successfully run './device-modbus -cp -d -o', I encounter 'access denied' error when performing read/write operations with Edgex and modbus-device-RTU. Even though I used 'sudo chown :2001 /dev/tty*' to address this, the error persists, as follows:
level=INFO ts=2023-12-19T01:50:54.196043628Z app=device-modbus source=modbusclient.go:42 msg="Modbus client create RTU connection."
level=INFO ts=2023-12-19T01:50:54.196183125Z app=device-modbus source=driver.go:116 msg="Read command OpenConnection failed. err:permission denied \n"
level=ERROR ts=2023-12-19T01:50:54.196351139Z app=device-modbus source=restrouter.go:159 X-Correlation-ID=a1edcfe2-cadc-4f48-b2c3-774203bb82f7 msg="error reading Regex DeviceResource(s) AISAMPLE for Modbus-RTU-IO-Module -> permission denied"

can you help me!
thanks a lot

jiekechoo, 2023-12-20:
@Liguangwei0321 , you can refer the documentation: https://docs.edgexfoundry.org/3.1/microservices/device/services/device-modbus/GettingStarted/#set-up-the-modbus-rtu-device

Liguangwei0321, 2023-12-20:
由于 没有用户组权限，使用sudo chmod a+rw /dev/ttyUSB0修改权限，成功解决该问题

closed 2023-12-20 by Liguangwei0321