edgexfoundry/device-modbus-go #550: Float32 rawType support

maxcarrel, 2024-04-27:
# 🚀 Feature Request

### Relevant Package
This feature request is for device-modbus-go

### Description
The documentation states that only Int16, Int32 and UInt16 rawType are supported at the moment. I would like to have Float32 supported as well. Looking at the codebase, it seems that the processing is actually already implemented: [deviceclient.go](https://github.com/edgexfoundry/device-modbus-go/blob/56822b146883d567dc10738adb0b8ce5338db960/internal/driver/deviceclient.go#L142). Is there a reason for this rawType to be filtered out in [utils.go](https://github.com/edgexfoundry/device-modbus-go/blob/56822b146883d567dc10738adb0b8ce5338db960/internal/driver/utils.go#L28)?

I am happy to provide a PR to fix this if there are no specific reason not to!

P.S: Thanks for the great work!

cloudxxx8, 2024-04-29:
the reason we need rawType because the reading values sent out might be calculated by `scale` and data type will become different from the Modbus Device
https://docs.edgexfoundry.org/3.1/microservices/device/services/device-modbus/details/DataTypeConversion/
In this example, it's integer to float.
If the valueType is the same as rawType, we don't need the additional rawType attribute.
In your case, I guess it would be rawType is float32, and value type is float64, right?
If the is the real case, please feel free to opne a PR.

maxcarrel, 2024-04-29:
Thank you for your quick reply. Yes, the need for rawType is quite clear to me. What I would need is rawType float32, value type float32. It seems that this is not supported at the moment but the required processing is already implemented. Are you aware of a reason for it not to be supported? If not I'll open a PR for this.

cloudxxx8, 2024-04-30:
@maxcarrel if your rawType and valueType are both float32, you can just define valueType without rawType.  Then, you don't need to make any code change.  I don't think we need additional process for this case.

maxcarrel, 2024-04-30:
You are right! Setting both rawType and valueType to float32 does not work but dropping rawType from the configuration does exactly what I want. I got a bit confused with this, my bad.

Thank you for the support!

closed 2024-04-30 by maxcarrel