edgexfoundry/device-modbus-go #629: Combine modbus register requests

sansmoraxz, 2025-03-27:
# 🚀 Feature Request

### Relevant Package [**REQUIRED**]

`edgexfoundry/device-sdk-go` - handles the device commands
`device-modbus-go` should combine modbus requests whenever possible while using `deviceCommands`

### Description [**REQUIRED**]
 A clear and concise description of the problem or missing capability...

Our RTU devices are facing quite heavy traffic and we have to use quite large timeout. It's a huge performance improvement if we can combine the requests. We do observe frequest requet failures while reading, even with large poll rates.

We have around 30 devices with around 16 device resouces on average over a single RTU port.

Take the following sample profile:-
```yaml
name: TEMP-HUMIDITY-SENSOR
manufacturer: Pluggable
description: "Plugable RS485"
model: PluggableModel
labels: [modbus]
deviceResources:
    - name: TEMPERATURE_VALUE
      isHidden: false
      properties:
        valueType: Int16
        readWrite: R
      attributes:
        primaryTable: HOLDING_REGISTERS
        rawType: Int16
        startingAddress: 0
    - name: HUMIDITY_VALUE
      isHidden: false
      properties:
        valueType: Int16
        readWrite: R
      attributes:
        primaryTable: HOLDING_REGISTERS
        rawType: Int16
        startingAddress: 1
deviceCommands:
  - name: "COMBINE_POLL"
    readWrite: "R"
    isHidden: false
    resourceOperations:
      - { deviceResource: "TEMPERATURE_VALUE" }
      - { deviceResource: "HUMIDITY_VALUE" }
```

`COMBINE_POLL` command edgex should fire a single command rather than two seperate commands as the registers are right next to each other. 

### Describe the solution you'd like
 

The following command in [modpoll](https://www.modbusdriver.com/modpoll.html) reads both values in a single command through viz. :-

```bash
modpoll -m rtu -b 9600 -a 1 -r 1 -c 2 -t 4 -p none /dev/ttyUSB0
```

Something like this may be implemented into the device-sdk-go to handle reading a sequence of registers in a single command.

Edgex may sort the resourceOperations by primaryTable and startingAddress and make combined requests for the addresses in sequence.

### Describe alternatives you've considered

Tried increasing the poll rate but it's not helping much. Same with having different poll rates for different device resources. Even implementing client side polling we face severe lag in the readings.
We realised that using combined requests will severly reduce the traffic on our devices.

sansmoraxz, 2025-03-27:
I guess this function may be optimized to be done smartly

https://github.com/edgexfoundry/device-modbus-go/blob/160951fd142c2a29a27987a636420e4249c6f47a/internal/driver/driver.go#L117-L153

biswadeepdey, 2025-04-18:
Facing the same issue as well.
I assume this package can do wonders if it supports Batch Read and Events for Connection and Disconnection.

I am yet to make changes for the batch read in the GO code as GO is new for me so I use CoPilot.

To handle the connection and disconnection, i have added custom event in driver.go. It is a major workaround but a required one.

cloudxxx8, 2025-04-21:
I can confirm it's totally doable