edgexfoundry/device-modbus-go #630: Unmaintained modbus library being used

sansmoraxz, 2025-03-31:
[goburrow/modbus](https://github.com/goburrow/modbus) is the library currently being used by edgex which have lot of unresolved issues. In particular we have observed https://github.com/goburrow/modbus/issues/35, https://github.com/goburrow/modbus/issues/54, https://github.com/goburrow/modbus/issues/64. (Happens with certain live devices not able to replicate with simulators though).

Also this library is not at all maintained (last commit was from 2018). In light of this might I suggest switching to a more active modbus library.

cloudxxx8, 2025-04-01:
We noticed that. This library was the only option when we started implementing this device service.
We are welcome for any contirbutors to help switch the library.

sansmoraxz, 2025-11-06:
What do you think of https://github.com/grid-x/modbus?

cloudxxx8, 2025-11-06:
we have already switched to it
https://github.com/edgexfoundry/device-modbus-go/blob/39b981e46d8fe8411fd729ef18898e222cbfbd49/go.mod#L8

It's also in the Odessa branch and we plan to release v4.0.1 this monthy
https://github.com/edgexfoundry/device-modbus-go/blob/ef6fb8718a92f9c9df7c8df871cfdac15af45b6b/go.mod#L8

cloudxxx8, 2025-11-06:
No, but you can open a PR for the v3 branch